martinberlin/cale-idf #7: Extended ASCII characters with TomThumb

BitSalat, 2020-06-17:
Really a great repo, I was able to compile and display on 7.5" Waveshare without issues. However would need extended ASCII for showing ° (deg) and Umlauts öüä etc. Tried with TomThunb but although setting extended to "1" it only shows chars up to 127. Additionally it is too small. Is there a way to get extended ASCII Fonts?

martinberlin, 2020-06-19:
Thanks @BitSalat happy that you like it. 
This part you have to check/read in Adafruit GFX that is the component that does the font. I also live in Germany and use umlauts.

Left a comment in main.cpp: 

// FONT used for title / message body - Only after display library
Converting fonts with ümlauts: ./fontconvert *.ttf 18 32 252

I use linux so no idea with other platforms. But basically you need to run this fontconvert tool and pass a bigger number so it takes all character spectrum. Be aware that of course more characters mean that the font will take also more DRAM from your device. 
Have fun with the library and spread the word if you like it, we need more testers!

BitSalat, 2020-06-19:
Thanx for your Response - we want to use it for a e-Paper Display connceting to an OpenHAB server, reading REST variables from there on a regular base and display them. Big Advantage would be to have a Display without any wiring as a battery inside should be enough to keep the Display working for several month.
So tested your repo adding on line with TomThum font invoked where we have seen that this is a full 256items font. However Problem seems to be that esp-idf does not handle Chars beyond 127 (for which reason ever). 
[image]
As you might see the line ends with "~" which is 0h7f, no further characters although we printed all 255 characters.

martinberlin, 2020-06-20:
Interesting. Going to try that myself and will leave some fonts (with “German” in the name) and try it out. But are you sure not any of the characters work?
Like Ö ü etc?

Good luck with your project. Next week I will be adding more classes to support more displays. For the 800*480 is already SPI optimized and should work faster than GxEPD

martinberlin, 2020-06-21:
Test with a full exported font with all 255 characters. Leave some a German fonts available in the process.
Reference: https://github.com/martinberlin/cale-idf/blob/refactor/oop/components/CalEPD/epd.cpp#L14 > write is the bridge to GFX 

It has a uint8 unsigned 8 bit value so it should support 255 characters. Not sure if the for ...auto does.

martinberlin, 2020-06-21:
Dear @BitSalat 
Please check my last commit in branch refactor/oop 
I generated Ubuntu font with all characters and printed it successfully in the Epaper. So I guess there should be something wrong with TomThumb font itself.

BitSalat, 2020-06-22:
Dear @martinberlin,
really impressive and "blitzsauber" as we say in German. Again thank you very much for your efforts showing me how 8-bit fonts can be managed. I will clone your branch and try it on my waveshare display. Also I will try to generate some fonts on my own and will report here how it has worked for me.

martinberlin, 2020-06-22:
Servus. Kein problem, viel Glück mit das testing. Also wenn du nur CalEPD als ESP-IDF component benutzen willst, kannst du direkt mit git submodules arbeiten und CalEPD + Adafruit GFX verwenden:

https://github.com/martinberlin/CalEPD - Just the ESP-IDF component (Note: Needs Adafruit GFX in order to use Fonts/Geometric functions)
https://github.com/martinberlin/Adafruit-GFX-Library-ESP-IDF Arbeitet zusammen mit CalEPD als GFX/Fonts

BitSalat, 2020-06-22:
Hi martinberlin, Danke für den Hinweis. Habe gerade refactor/oop mit Eclipse compiliert und auf mein System geflasht. Compilierte sofort und war mit dem vollständigen Font auf dem Display. Klasse!
Probleme habe ich allerdings mit dem fontconvert, gerade MINGW installiert und versucht, die Freetype library zu bilden - keine Chance, steigt aus mit Fehlermeldung zlib.h nicht vorhanden, obwohl definitiv da. Readmes zum MinGW library path sind beliebig kryptisch.
Gibt es kein einsatzfähiges Utility (egal ob Win oder Linux) mit dem sich die Fonts generieren lassen?
Grüsse, BitSalat

BitSalat, 2020-06-22:
Hier noch ein Foto meines Displays:
[image]

martinberlin, 2020-06-22:
Also ich benutzte Ubuntu, und hatte kein Problem, fontconvert zu kompilieren. Ich habe das kompilierte fontconvert dort liegen lassen aber klar braucht Freetype lib....
Ich mag auch der ESP-IDF toolchain, kompiliert viel schneller als wenn ich Sachen mit Platformio/arduino-esp32 entwickle. Deswegen für zukünftige Firmware wurde ich nur versuchen ESP-IDF im Griff nehmen.

Komisch das lass dich das nicht installieren.  Aber sag mir einfach welche Fonts du brauchst oder schickt mir die TTF und ich generieren eine /german Verzeichniss mit alle characteren. 

Wenn du Zeit hast und diese repository interessant findest bitte lass eine Sternchen ;)

BitSalat, 2020-06-22:
Klar, volle Punkt- und Sternzahl - wusste gar nicht, dass sich das in Github machen lässt, habe den Star-Button aber doch gefunden ...
Ja mit den Fonts das wär natürlich Klasse - der Ubuntu Font ist geradlinig, serifenlos und kommt sehr sauber auf dem S/W Display `rüber - also eine Handvoll davon mit 8, 12, 16, 20 und 24pt wären die Grössen, die wir wir brauchen. Dann könnten wir das Thema ePaper tatsächlich abschließen und uns unserer OpenHAB REST-API widmen.
Das mit CALE.es probier ich gerne aus, werde mich mal (vorauss. morgen) registrieren.
Nochmal Danke für Deinen tollen Support!
P.S: Mein Kompagnion hat gefragt, ob Du das Waveshare Gehäuse verwendest? Der batzelt nämlich gerade mit einem 3D-Drucker herum (ich habe ein Standard OKW Gehäuse verwendet).

martinberlin, 2020-06-22:
Kein Problem dann morgen oder sogar heute Abend füge ich: 8, 12, 16, 20 und 24pt hinzu
Evtl. mache ich eine Fonts/ubuntu neue Verzeichnis damit ist nicht mit dem Originals gemischt. Antwort für deine Kollege:
Ich habe meine eigene Back-Housing mit Blender und die 3D-Drücker gemacht: https://www.thingiverse.com/thing:4126236

Aber das Front-Housing ist von Waveshare. Was ich cool finde ist das die Front sieht schoner wenn ist nicht 3D-gedrückt :)

martinberlin, 2020-06-22:
[image]
Check my last commit. Fonts sind da!
Sag mir Bescheid wenn das funktioniert wie erwünscht.

BitSalat, 2020-06-24:
Habe Dein repo gezogen und compiliert, alle Fonts funktionieren bestens, nochmals herzlichen Dank!

martinberlin, 2020-06-28:
Kein problem, freue mich darauf, stolz zu sehen diese Component ist benutzbar für Firmwares!
Viel Spaß damit und keep in touch

BitSalat, 2020-07-02:
Hi Martin,
leider gibt es noch ein Problem mit den Umlauten, hier ein einfacher Text zum Test

   display.setFont(&Ubuntu_M8pt8b);
   display.print("Ubuntu 8pt");
   display.println("");
   **display.print("Test-Text: Öfter mal was ändern! Vor allem die Temperatur °C");**

Dabei erscheint auf dem Display das:

[image]

Characters oberhalb von 0x7f werden offensichtlich durch 16 bit UTF-8 repräsentiert und erscheinen dann als doppeltes Zeichen (aber falsch codiert) auf dem Display.
Ich bin etwas ratlos, wie man das lösen kann ...

martinberlin, 2020-07-02:
Pushed a small fix to refactor/oop
UMLAUTEN Issue recherche
display.println("äöü ÄÖÜ");

// Is trying to print 2 chars per letter (Which is weird, like if it was a multi-character)

char: 195
char: 164
char: 195
char: 182
char: 195
char: 188
char: 32
char: 195
char: 132
char: 195
char: 150
char: 195
char: 156

So I just skipped 195 and changed the others to the right characters. Did this only on println in epd.cpp as an example!
Hope you find it useful. Nice catch this needs more research

BitSalat, 2020-07-02:
Hi Martin,
thanx for your fast reaction - I think it has something to do with character handling, probably we'll need a conversion from wchar_t to char (iconv lib?) so from Unicode to ASCII.
We found the issue when fetching REST variables from an OH server with the simple string "25.5 °C". The first 4 positions (25.5 ) where coded as normal ASCII however for the "°" we got a double byte ASCII 0xb2c0 (looks like UTF-8)
The cale display took this as 2 chars, one is a special A (with something on top) the other indeed is °
So it's probably necessary to have a type conversion, usually iconv does this job but I do not know if it's implemented in ESP-IDF.
BitSalat

martinberlin, 2020-07-02:
Yes that's it. I'm now in Urlaub for some days. But following my example we can handle this also for the degrees Sign. 
Out of range characters are coming as a double uint8 starting with 195 so the trick I found is to skip this one and replace it by the right Adafruit character number. Will probably find some minutes tonight to make this replace into a function and then when I come back we research the unicode to ASCII method. 
For sure there is a smarter way to do it for all characters at once.

BitSalat, 2020-07-02:
Hi Martin,
enjoy your Urlaub!
BitSalat

martinberlin, 2020-07-02:
Sure. Still not flying. Check this one and compare it with my update in epd.cpp
https://www.rapidtables.com/code/text/unicode-characters.html
[image]
That is Ä which is updated here:
https://github.com/martinberlin/cale-idf/blob/refactor/oop/components/CalEPD/epd.cpp#L37

So we just need to add the most used characters there and it will work even without unicode library. 

º is 186

martinberlin, 2020-07-02:
I just pushed an update that seems to solve this issue. Feel free to test and if it works send me a message.
Adding complete unicode support like that fork of Adafruit GFX I referenced does not make too much sense at this point, since I'm not planning to add Asian languages (It also stops the setFont functionality in support for BMP fonts so it's a no go)
But I found a way to replace the characters. Left two options in Epd.cpp and the one selected as default seems to work fine. Small preview:
[image]

martinberlin, 2020-07-12:
Hallo @BitSalat 
Hat das funktioniert?
Morgen bin ich zurück in Hauptstadt :)

BitSalat, 2020-07-13:
Yes, worked, we can now use the German Umlauts and special characters.
We have already implemented but now see an issue with display.update - every update causes 3 transistions from black->white before new contents is visible. As we are using the waveshare 7.5" V2 display I assume that a partial update is not possible?
However why do we need 3 transistions, is there a way to reduce to one transition?

martinberlin, 2020-07-13:
The 3 transitions seem standard in most good display epapers on most sizes (Waveshare uses good display with added electronics or their own SPI interface too AFAIK)
Still need to debug and find out why partial update is not working as it should. Will dig later the documentation and see what I find out.

martinberlin, 2020-07-13:
This issue is now closed since the original request is resolved. Let's open new tickets so it's easier to focus in the task that needs to be targeted. Thanks for all your testing @BitSalat

closed 2020-07-13 by martinberlin

sektion4, 2025-08-30:
Hi,
I really like the fonts because they include extended characters, which I use frequently in my projects. Would it be possible to get the Ubuntu_M36pt7b font with extended characters as well?

martinberlin, 2025-09-01:
@sektion4 this is issue is closed and won't be reopened for this. Please don't be lazy and read about fontconvert, find a way to do this yourself. Thanks!