martinberlin/cale-idf #10: Reduce black-white transistions during display update or invoke partial update

BitSalat, 2020-07-14:
We have implemented latest cale-idf repo but now see an issue with display.update - every update causes 3 transistions from black->white before new contents is visible. As we are using the waveshare 7.5" V2 display I assume that a partial update is not possible?
However why do we need 3 transistions, is there a way to reduce to one transition?

martinberlin, 2020-07-15:
The only way I know is doing a partial update. This uses only one transition. But on 7.5" V2 is still not working like it should. We will need to go together through the documentation.
On GxEPD is also not working so we may be missing some SPI command or there is something that needs correction. Let's start by putting links to the available documentation:
https://www.waveshare.com/wiki/7.5inch_e-Paper_HAT_(B) is quite confusing. Better check the Good display documentation. My presumption is that Waveshare only adds electronics on top, but EPD's are good display

GD7965 is the controller. In http://www.good-display.com is not listed anymore so we may have to email them. It's not easy to find documentation for this ones, since sometimes they remove models if there are not available anymore.

[Wiki link of 7.5 V2](https://github.com/martinberlin/cale-idf/wiki/Model-gdew075T7.h)

BitSalat, 2020-07-15:
Will check if I can find a datasheet later - meanwhile do you know the Zephyr project: [https://www.zephyrproject.org/](https://www.zephyrproject.org/)?
Basically this is a RTOS for embedded devices but they do have display drivers and are also supporting the GD7965 here:
[https://github.com/zephyrproject-rtos/zephyr/blob/7a822078fcac21012d5cbc498f469fa2a81895d6/drivers/display/gd7965.c](url)
Maybe something we can get some inputs ... 
BTW you can still find the GD7965 datasheet here: [https://v4.cecdn.yun300.cn/100001_1909185148/GD7965.pdf](url)

martinberlin, 2020-07-15:
Nice will research a bit tonight and check against the commands we are sending. I hope it can be fixed since I'm also very interested in get partial updates working as they should

martinberlin, 2020-07-16:
Will be some days away from the keyboard since I’m going to Poland. Will keep this one on my mind for further research

martinberlin, 2020-07-19:
Ref and sample code from Waveshare:
https://github.com/waveshare/e-Paper/blob/master/Arduino/epd7in5_V2/epd7in5_V2.cpp

martinberlin, 2020-07-21:
@BitSalat Added a new class wave7i5.h 

Still not tested, have to coordinate with my father, since I do not have this EPD at home anymore (Was a gift to him)

BitSalat, 2020-07-22:
Dev boards are on their way to you - please check your email!

martinberlin, 2020-07-22:
Many thanks. Sadly I have 2 not so good news:

1. 800*480 seems to be out-of-stock in Waveshare
[image]

2. Even trying with [Waveshare 7.5 EPD example](https://github.com/waveshare/e-Paper/blob/master/Arduino/epd7in5_V2/epd7in5_V2.cpp#L133), I cannot do a successful partial refresh: 

```c++
void Epd::Displaypart(const unsigned char* pbuffer, unsigned long xStart, unsigned long yStart,unsigned long Picture_Width,unsigned long Picture_Height) {
    SendCommand(0x13);
    int * padd;
    // xStart = xStart/8;
    // xStart = xStart*8;
    for (unsigned long j = 0; j < height; j++) {
        for (unsigned long i = 0; i < width/8; i++) {
            if( (j>=yStart) && (j<yStart+Picture_Height) && (i*8>=xStart) && (i*8<xStart+Picture_Width)){
                SendData(~(pgm_read_byte(&(pbuffer[i-xStart/8 + (Picture_Width)/8*(j-yStart)]))) );
                // SendData(0xff);
            }else {
                SendData(0x00);
            }
        }
    }
    SendCommand(0x12);
}
```

This in my opinion is just like a normal update. See that in the other class is sending a "Partial IN" command and some Ram pointers, to signalize the update area. 
Like this, unless we find the right commands, in a documentation that is well written will be really hard to get this working.

martinberlin, 2020-07-22:
@BitSalat sent an email to good display about partial update and added here:
https://github.com/martinberlin/cale-idf/wiki/Model-gdew075T7.h -> Link to the IC controller PDF (In page 10 the partial update reference)

martinberlin, 2020-07-29:
@BitSalat latest news: I ordered last week a "7,5 zoll bunte e-ink-display mit 800x480 auflösung epaper display" that has same IC controller as this one. So I will be able to test it again at home, let's see if we discover something. I will also resend the support email to Good Display
[image]

martinberlin, 2020-07-30:
Last 7.5 I ordered does not support partial update. News: Good Display sent an example program that performs a partial update (In other platform)
[GDEW075T7_P_20191125.zip](https://github.com/martinberlin/cale-idf/files/4999870/GDEW075T7_P_20191125.zip)

BitSalat, 2020-07-31:
thanx for your continued efforts for the partial update however I understand that it is not supported on the 7.5" displays from Waveshare or GoodDisplay. Perhaps there will be new devices in the future.

martinberlin, 2020-07-31:
It's not supported on the color ones. In 7.5" 2 color is supported and I attached the Good Display example.
There is not such a thing as Waveshare EPD, their displays are manufactured by Good Display, they add external electronics and rebrand them as Waveshare but the one that really has last word about what the epaper can or cannot do is Good display.
So I'm attaching their example, we just have to try hard to adapt it, but it may work.

martinberlin, 2020-08-14:
Dear Stefan I will try to get one 7.5 epaper again this time. Tried last week but they are out of stock. This epaper have support for partial update but is not easy to do and undocumented.
Also the gxEPD example does not work so it’s not an easy thing to get working.
I will also implement an additional buffer for light gray, this will take an additional:
Width*height/8 bytes buffer (each byte stores 8 monochrome pixels)

martinberlin, 2020-10-10:
New 7.5 gdew075T7 was purchased days ago. When it arrives I will make a deep test on this one again and try to solve the partial update and find a way to make the temperature issue better (If possible)
So leaving this open till then, sorry for the long wait!!!
Please try the touch if you can and in next purchase get a cheap 2.7 touch epaper from Good Display (15 u$ aprox.)

martinberlin, 2021-01-04:
Such a bad luck with this one. It arrived but didn't left any message, so it comes back to China...AHhhm. And after they send again I will finally get it ;)

martinberlin, 2021-07-09:
@BitSalat my new 7.5 epaper is finally coming to Berlin in the next weeks. Sorry that it took so long. I’m still very grateful to use the ESP32 controllers you sent me and about the growth of this library because of you, the early main adopters

BitSalat, 2021-07-12:
@martinberlin thanx for info. We are using your cale-idf for our OpenHAB / HomeAssistant ePaper displays with great success and wouldn't have been able to do so without your libraries. There's only one thing still a bit annoying. For updating the screen it takes 4 changes between black and white which looks not so sexy. Is there a way to reduce that?

martinberlin, 2021-07-12:
There is partial update. But is not as clean as full update. When this 7.5 arrives will make a branch so we can test it together. 
I guess a good combination would be to do partial refresh and from time to time a full refresh. Full always makes that 4 flashes. It seems the only way to fully clean the Eink from the screen is doing that. Will get back to you soon

martinberlin, 2021-07-30:
EPD 7.5 arrived, will leave it wired to SPI to try new partial display, Goodisplay sent this function, it's for SMT32 but what matters is the CMD and data sent:

```c
/***************************partial display function*************************************/

void partial_display(u16 x_start,u16 x_end,u16 y_start,u16 y_end ,const unsigned char *old_data,const unsigned char *new_data,unsigned char mode) //partial display
{
   unsigned int i;
		EPD_W21_WriteCMD(0X50);
		EPD_W21_WriteDATA(0x47);		
		lut1();
	  EPD_W21_WriteCMD(0x91);		//This command makes the display enter partial mode
		EPD_W21_WriteCMD(0x90);		//resolution setting
		EPD_W21_WriteDATA (x_start/256);
		EPD_W21_WriteDATA (x_start%256);   //x-start    		
		EPD_W21_WriteDATA (x_end/256);		
		EPD_W21_WriteDATA (x_end%256-1);  //x-end

		EPD_W21_WriteDATA (y_start/256);
		EPD_W21_WriteDATA (y_start%256);   //y-start    	
		EPD_W21_WriteDATA (y_end/256);		
		EPD_W21_WriteDATA (y_end%256-1);  //y-end

		EPD_W21_WriteCMD(0x10);	       //writes Old data to SRAM for programming
 
	if(mode==0)
	for(i=0;i<48000;i++)	     
	{
	 EPD_W21_WriteDATA(0xff);  
	}
	else
	 for(i=0;i<48000;i++)	     
	{
	 EPD_W21_WriteDATA(~old_data[i]);  
	}  
	
	EPD_W21_WriteCMD(0x13);				 //writes New data to SRAM.
		for(i=0;i<48000;i++)	     
	 {
		EPD_W21_WriteDATA(~new_data[i]);  
	 }  
	 
		EPD_W21_WriteCMD(0x12);		 //DISPLAY REFRESH 		             
		driver_delay_xms(300);     //!!!The delay here is necessary, 200uS at least!!!     
		lcd_chkstatus();
	  
}
```

- [ ] Fix partial update
- [ ] Check how to add 4 grays support

martinberlin, 2021-07-30:
@BitSalat just corrected the partial update Waveform in branch: **refactor/oop**

You can try to use partial update now. Try flashing demo-epaper.cpp adding it in the CMakeFiles and commenting the rest. 

Later I will upload a short video

martinberlin, 2021-11-26:
Dear @BitSalat closing this since I didn't got any feedback since 30 July (4 months aprox.)
Feel free to re-open if there is something more to do.

closed 2021-11-26 by martinberlin