martinberlin/cale-idf #18: Implement 11.6 inch 960x640 resolution black/red, GDEH116Z91

martinberlin, 2020-11-15:
[Sample code](https://www.good-display.com/companyfile/375.html)

ATC take: https://github.com/atc1441/E-Paper_Pricetags/blob/main/GxEPD2_modded/src/epd3c/GxEPD2_1160c.cpp

martinberlin, 2020-11-15:
esptool.py v3.0-dev
Generated /home/martin/esp/projects/cale-idf/build/cale-epd.bin

Replace /home/martin with your own path to home. Here are the bin files:
[atc.zip](https://github.com/martinberlin/cale-idf/files/5543019/atc.zip)

Project build complete. To flash, run this command:
/home/martin/.espressif/python_env/idf4.2_py3.8_env/bin/python ../../esp-idf/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size detect --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/cale-epd.bin
or run 'idf.py -p (PORT) flash'

atc1441, 2020-11-15:
Tested the firmware but it does not boot, 

the watchdog gets triggered

martinberlin, 2020-11-16:
Mh strange. Watchdog usually does trigger after boot, and usually you get the alert and keeps on working.
Tell me when you have time and I will setup the toolchain for you in Ubuntu

adrien3d, 2020-11-21:
Same here looping after `frequency: 4000000`

```
CalEPD component version 0.9.8
Gdeh116Z91() constructor injects IO and extends Adafruit_GFX(960,640) Pix Buffer[76800]

Available heap after Epd bootstrap:30768
I (370) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
CalEPD version: 0.9.8
Fonts-demo. Free heap: 143108 (After epaper instantiation)
DRAM     : 143108
Gdeh116Z91::init(debug:1)
MOSI: 23 CLK: 18
SPI_CS: 32 DC: 27 RST: 26 BUSY: 35

EpdSPI started at frequency: 4000000
E (10380) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10380) task_wdt:  - IDLE (CPU 0)
E (10380) task_wdt: Tasks currently running:
E (10380) task_wdt: CPU 0: main
E (10380) task_wdt: CPU 1: IDLE
E (10380) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x400D8E00:0x3FFB0810 0x400823AD:0x3FFB0830 0x400D5732:0x3FFDAD60 0x400D5A03:0x3FFDAD80 0x400D54D3:0x3FFDADA0 0x400E9954:0x3FFDAEB0 0x40087CED:0x3FFDAED0 
0x400d8e00: task_wdt_isr at /Users/adrien/Dev/esp/esp-idf/components/esp_common/src/task_wdt.c:187

0x400823ad: _xt_lowint1 at /Users/adrien/Dev/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x400d5732: Gdeh116Z91::fillScreen(unsigned short) at /Users/adrien/Dev/epaper/cale-idf/build/../components/CalEPD/models/color/gdeh116Z91.cpp:39

0x400d5a03: Gdeh116Z91::init(bool) at /Users/adrien/Dev/epaper/cale-idf/build/../components/CalEPD/models/color/gdeh116Z91.cpp:27

0x400d54d3: app_main at /Users/adrien/Dev/epaper/cale-idf/build/../main/demos/demo-fonts.cpp:65

0x400e9954: main_task at /Users/adrien/Dev/esp/esp-idf/components/freertos/port/port_common.c:133 (discriminator 2)

0x40087ced: vPortTaskWrapper at /Users/adrien/Dev/esp/esp-idf/components/freertos/port/xtensa/port.c:168

E (10380) task_wdt: Print CPU 1 backtrace
```

martinberlin, 2020-11-21:
Thanks Adrien, 
I don't have this epaper Model to test that is the problem. So I guess the best move here would be to install ESP-IDF to @atc1441
and I can then connect and debug remotely.

Other than that, if you find what it is, feel free to make a Merge request and I will merge it so it's available for the rest. Thanks a lot for trying CALE-IDF is the first time I make a component for the Espressif Framework

martinberlin, 2020-11-21:
Other than that @atc1441 this called my attention:
Available heap after Epd bootstrap:30768  -> That is around 30 Kb Heap. Can be that the 960*640*2 is eating all the RAM?

GFX  RAM Usage (Black,red):
960*640/8  = 76800  also 76 Kb * 2 = 152 Kb approx. to handle GFX. That's a bit too much. Maybe that's a reason why it hangs? We are just eating with our huge GFX array all the available RAM in ESP32? 
Maybe it's totally different. Should be since you got it working :)

atc1441, 2020-11-21:
@martinberlin  couldnt you test the bare .bin on your system till it runs, even without the display it should boot for you. Just thinking about that

martinberlin, 2020-11-23:
Hey I discovered what it was, the fillscreen() Method was using a uint16 unsigned, that holds max 65535, that generated a never ending dead loop since:

for (uint16_t x = 0; x < sizeof(_buffer); x++)  

x was never going to arrive to 76800 that is the buffer size per color in this display. @adrien3d can you try it?
If it works for you then I will generate a new bin for Aaron. But just tell me one afternoon and I install you IDF, since you will need it to test the touch epaper I'm sending you soon.

adrien3d, 2020-11-24:
Ok, so the good new is that it is actually doing something (although I noticed I had to connect the FPC the other way around). But it just shows a noisy picture.
[image]

With my code (just a straightforward implementation of atc's): https://github.com/adrien3d/gdeh116Z91-test/blob/master/main/epaper_main.c, I made it "working" by putting `0xF7` instead of atc `0xC7` for register `0x22`.
[image]

I am trying few things (more inspired by good-display code) and will keep you updated in case of progress.

martinberlin, 2020-11-24:
Great but at least does something. Let's wait for @atc1441 feedback for sure we will get it working soon :)
Feel free to modify the commands and check them with Aaron example. It's very possible that I forgot something or there is a command missing. I'm also not sure if it's better to do a software reset or a hardware reset (With the RST pin)
Anyways I'm a bit disabled since I can compile it on an ESP32 but I don't have any test epaper so it's hard to debug it.

martinberlin, 2021-01-16:
Hey @atc1441 and @adrien3d any interest to close this one?

If it does not work just point me where to get one of this or send me one. But I would like to either implement it to the end or discard it if there is no interest and close this issue

atc1441, 2021-01-16:
So far it can be closed from my side so you dont habe the opened issue here, when the time has come i will try it out :)

martinberlin, 2021-01-16:
No problem leaving it on hold will close at the end of this month. Thanks

closed 2021-01-29 by martinberlin