martinberlin/cale-idf #21: unable to build

xeijin, 2020-11-25:
Hello, thank you for your work here.

I am trying to build the project from master (using platformIO from within VSCode).

Unfortunately I am hitting this error and unsure why.

Hardware: TTGO T-Beam ESP32, Waveshare 12.48in epaper module B

```sh
> Executing task in folder cale-idf: pio run <

Warning! Ignore unknown configuration option `monitor_filters` in section [platformio]
Processing ttgo-t-beam (platform: espressif32; framework: espidf; board: ttgo-t-beam)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/ttgo-t-beam.html
PLATFORM: Espressif 32 (2.0.0) > TTGO T-Beam
HARDWARE: ESP32 240MHz, 1.25MB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-espidf 3.40100.200827 (4.1.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - tool-ninja 1.9.0 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80200.200827 (8.2.0)
Reading CMake configuration...
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/ttgo-t-beam/esp-idf/main/cale.cpp.o
Archiving .pio/build/ttgo-t-beam/esp-idf/app_trace/libapp_trace.a
Indexing .pio/build/ttgo-t-beam/esp-idf/app_trace/libapp_trace.a
Archiving .pio/build/ttgo-t-beam/esp-idf/app_update/libapp_update.a
Indexing .pio/build/ttgo-t-beam/esp-idf/app_update/libapp_update.a
main/cale.cpp: In function 'esp_err_t _http_event_handler(esp_http_client_event_t*)':
main/cale.cpp:197:28: warning: comparison of unsigned expression < 0 is always false [-Wtype-limits]
             if (bmp.height < 0)
                 ~~~~~~~~~~~^~~
main/cale.cpp: In function 'void http_post()':
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::host' [-Wmissing-field-initializers]
         };
         ^
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::port' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::username' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::password' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::auth_type' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::path' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::query' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::cert_pem' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::client_cert_pem' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::client_key_pem' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::disable_auto_redirect' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::max_redirection_count' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::transport_type' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::buffer_size_tx' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::user_data' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::is_async' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::use_global_ca_store' [-Wmissing-field-initializers]
main/cale.cpp:407:9: warning: missing initializer for member 'esp_http_client_config_t::skip_cert_common_name_check' [-Wmissing-field-initializers]
main/cale.cpp: In function 'void wifi_init_sta()':
main/cale.cpp:488:5: error: 'esp_event_handler_instance_t' was not declared in this scope
     esp_event_handler_instance_t instance_any_id;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
main/cale.cpp:488:5: note: suggested alternative: 'esp_event_handler_unregister'
     esp_event_handler_instance_t instance_any_id;
     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 63 lines cut ...]
                               ^
main/cale.cpp:538:89: error: 'instance_any_id' was not declared in this scope
     ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
                                                                                         ^~~~~~~~~~~~~~~
/Users/hassan/.platformio/packages/framework-espidf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
main/cale.cpp:538:21: error: 'esp_event_handler_instance_unregister' was not declared in this scope
     ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/hassan/.platformio/packages/framework-espidf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
main/cale.cpp:538:21: note: suggested alternative: 'esp_event_handler_unregister'
     ESP_ERROR_CHECK(esp_event_handler_instance_unregister(WIFI_EVENT, ESP_EVENT_ANY_ID, instance_any_id));
                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/hassan/.platformio/packages/framework-espidf/components/esp_common/include/esp_err.h:118:31: note: in definition of macro 'ESP_ERROR_CHECK'
         esp_err_t __err_rc = (x);                                       \
                               ^
Compiling .pio/build/ttgo-t-beam/esp-idf/asio/asio/asio/src/asio.cpp.o
*** [.pio/build/ttgo-t-beam/esp-idf/main/cale.cpp.o] Error 1
============================================================= [FAILED] Took 7.73 seconds =============================================================
The terminal process "pio 'run'" terminated with exit code: 1.

```

xeijin, 2020-11-26:
If I force it to use IDF v4.0.0 as suggested in `platformio.ini` it gives me this error on build:

`#include "esp_netif.h"`

but it looks like this include only became available from IDF v4.1?

closed 2020-11-26 by xeijin

martinberlin, 2020-11-27:
Great I will update the Readme.
Did it work for you?

Can you post some picture?
Thanks for trying it out

xeijin, 2020-11-28:
@martinberlin unfortunately not, I am using the colour version of the 12.48in waveshare display - let me open a separate issue to highlight where I got stuck.

martinberlin, 2020-11-30:
I implemented only the black and white so far since I never got this model to test it. 
But if you help me out we can get it working.