martinberlin/cale-idf #22: issue with 12.48in waveshare 3-colour

xeijin, 2020-11-28:
Hello, many thanks for your hard work on this, please note this is my first hardware project so appreciate your patience with me!

After a couple days hacking I have been unable to get [**_three colour_** version 12.48in waveshare display](https://www.waveshare.com/wiki/12.48inch_e-Paper_Module_(B)#ESP32) working.

**problem: epaper is not updating** (no effect on display with either `cale-idf` or the original waveshare demos)

----

##  board 
  - [LILYGO TTGO T-Energy (WROVER chip with PSIRAM)](https://www.amazon.co.uk/ICQUANZX-Bluetooth-Battery-ESP32-WROVER-B-Development/dp/B07VQJ1RH2/ref=pd_sbs_200_1/262-0163938-5202069?_encoding=UTF8&pd_rd_i=B07VQJ1RH2&pd_rd_r=159869bc-04d2-42e8-b8e7-938176466410&pd_rd_w=DVdhv&pd_rd_wg=wBfeU&pf_rd_p=2304238d-df78-4b25-a9a0-b27dc7bd722e&pf_rd_r=8SWY3JN2K2P6TQJPVX2Y&psc=1&refRID=8SWY3JN2K2P6TQJPVX2Y)
    - **the physical pin layout matches the WeMos D32 that you used** (and the waveshare board), though I think maybe the pinout differs
    - [manufacturer schematic](https://raw.githubusercontent.com/LilyGO/LILYGO-T-Energy/master/t18_v3.0.pdf)
    - [manufacturer pinout diagram](https://raw.githubusercontent.com/LilyGO/LILYGO-T-Energy/master/T18v3.0.jpg) 
      - **caution:** it looks like there are some mistakes in the diagram

   - **some wiring changes I had to make** vs your 1248in waveshare wiki page (the board does not have soldered headers, so I am using dupont jumper wires, I had to use slightly different pins to make this work)
      [image]

##  code
<details>
<summary><b>display configuration in <tt>sdkconfig</tt></b></summary>

```
# Display Configuration
#...
CONFIG_EINK_SPI_MOSI=14
CONFIG_EINK_SPI_CLK=13
CONFIG_EINK_SPI_CS=-1
CONFIG_EINK_DC=-1
CONFIG_EINK_RST=-1
CONFIG_EINK_BUSY=-1
#...
CONFIG_EINK_SPI_CS2=-1
CONFIG_EINK_SPI_MISO=-1
#...
CONFIG_EINK_SPI_M1_CS=23
CONFIG_EINK_SPI_S1_CS=22
CONFIG_EINK_SPI_M2_CS=26
# 16 -> 26
CONFIG_EINK_SPI_S2_CS=19
CONFIG_EINK_SPI_M1_BUSY=32
CONFIG_EINK_SPI_S1_BUSY=21
CONFIG_EINK_SPI_M2_BUSY=18
CONFIG_EINK_SPI_S2_BUSY=4
CONFIG_EINK_M1S1_DC=25
CONFIG_EINK_M2S2_DC=27
# 17 -> 27
CONFIG_EINK_M1S1_RST=33
CONFIG_EINK_M2S2_RST=5
# end of Display Configuration
```
</details>

<details>
<summary><b>serial output (with <tt>main/cale.cpp</tt>)</b></summary>

```
--- idf_monitor on /dev/cu.usbserial-0200D385 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7080
load:0x40078000,len:13964
load:0x40080400,len:4816
0x40080400: _init at ??:?
[... 172 lines cut ...]

I (9807) CALE: 130 len:1024

I (10167) CALE: 140 len:1024

I (10517) CALE: 150 len:1024

I (10877) CALE: HTTP_EVENT_ON_FINISH
Download took: 5763 ms
Refresh and go to sleep 60 minutes

Resolution setting
Sending a buffer[160392] via SPI

Available heap after Epd update: 4182592 bytes
STATS (ms)
888 _wakeUp settings+send Buffer
696 _powerOn
1584 total time in millis
```
</details>
  - my current fork: [xeijin/cale-idf (branch: 1248in-3colour)](https://github.com/xeijin/cale-idf/tree/1248in-3colour)
  - have also tried with `demo-fonts.cpp`, `demo-sleep-clock.cpp` etc. same issue with all

thank you in advance for any help!

martinberlin, 2020-11-30:
I will check your wiring tomorrow. But as mentioned in the other ticket, first I tried this with an ESP32-WROOVER tB hat is working and has the same pinout:
https://github.com/martinberlin/cale-idf/wiki/Model-wave12i48.h#esp32-wroover-hack-for-this-epaper

Other than that it might be that the Colour one has a different initialization but still it sounds to me that if nothing works it must be some wiring issue here. Because at least the Black /White is proved to be working, is on my wall since long time with the hack I described. But if the one you have does not update also with Waveshare example there must be something else. 

Double check it make sure the epaper has the 3.3v and all the full wiring connected with a tester tool. Another option to discard there is any issue with the epaper itself is to get an ESP32 Epaper driver: https://www.waveshare.com/wiki/E-Paper_ESP32_Driver_Board
And just flash it with the original example of Waveshare first. 
This is [my wiring](https://github.com/martinberlin/cale-idf/blob/master/config-examples/sdkconfig_Wave12I48_PSRAM_active) for the one that is proved to be working with the hack described before.

martinberlin, 2020-11-30:
Xeixin,
Please pull this branch: https://github.com/martinberlin/cale-idf/tree/epaper/22

And just build and flash monitor.  It has no color support but it's now sending additional LUT tables and different commands for this model

xeijin, 2020-12-01:
@martinberlin some strange results

At first it worked partially (and I actually have video proof! https://streamable.com/xc0jzt) - I can just barely make out some german text and only the top-right screen updated.

However I then flashed something else and tried flashing back and now getting this

<details>
<summary>**serial output**</summary>

```sh
--- idf_monitor on /dev/cu.usbserial-0200D385 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7084
load:0x40078000,len:13652
load:0x40080400,len:4632
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-rc-5-g511965b26-dirty 2nd stage bootloader
I (29) boot: compile time 00:18:30
I (29) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x259b8 (154040) map
I (178) esp_image: segment 1: paddr=0x000359e0 vaddr=0x3ffb0000 size=0x03a04 ( 14852) load
I (185) esp_image: segment 2: paddr=0x000393ec vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /Users/hassan/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (186) esp_image: segment 3: paddr=0x000397f8 vaddr=0x40080404 size=0x06820 ( 26656) load
I (206) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xa1474 (660596) map
0x400d0020: _stext at ??:?

I (482) esp_image: segment 5: paddr=0x000e149c vaddr=0x40086c24 size=0x0d830 ( 55344) load
0x40086c24: uninstall_priv_desc at /Users/hassan/esp-idf/components/driver/spi_master.c:706

I (509) esp_image: segment 6: paddr=0x000eecd4 vaddr=0x400c0000 size=0x00068 (   104) load
I (521) boot: Loaded app from partition at offset 0x10000
I (521) boot: Disabling RNG early entropy source...
I (521) psram: This chip is ESP32-D0WD
I (527) spiram: Found 64MBit SPI RAM device
I (530) spiram: SPI RAM mode: flash 40m sram 40m
I (536) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (543) cpu_start: Pro cpu up.
I (547) cpu_start: Application information:
I (551) cpu_start: Project name:     cale-epd
I (556) cpu_start: App version:      v0.9.8-59-g5b3e73a
I (562) cpu_start: Compile time:     Dec  1 2020 00:27:06
I (568) cpu_start: ELF file SHA256:  79288072e7b9b68f...
I (574) cpu_start: ESP-IDF:          v4.2-rc-5-g511965b26-dirty
I (581) cpu_start: Starting app cpu, entry point is 0x40081e98
0x40081e98: call_start_cpu1 at /Users/hassan/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
[... 247 lines cut ...]
E (15239) gpio: gpio_set_level(215): GPIO output gpio_num error
Resolution setting
E (15249) gpio: gpio_set_level(215): GPIO output gpio_num error
E (15259) gpio: gpio_set_level(215): GPIO output gpio_num error
Flash mode
E (15259) gpio: gpio_set_level(215): GPIO output gpio_num error
E (15269) gpio: gpio_set_level(215): GPIO output gpio_num error
E (15279) gpio: gpio_set_level(215): GPIO output gpio_num error
E (15279) gpio: gpio_set_level(215): GPIO output gpio_num error
E (17299) gpio: gpio_set_level(215): GPIO output gpio_num error
E (17299) gpio: gpio_set_level(215): GPIO output gpio_num error
Sending a 33600 bytes buffer via SPI
E (21379) gpio: gpio_set_level(215): GPIO output gpio_num error
E (21379) gpio: gpio_set_level(215): GPIO output gpio_num error

STATS (ms)
6255 _wakeUp settings+send Buffer
2000 update 
8256 total time in millis
E (23389) gpio: gpio_set_level(215): GPIO output gpio_num error
E (23389) gpio: gpio_set_level(215): GPIO output gpio_num error
E (25409) gpio: gpio_set_level(215): GPIO output gpio_num error
E (25409) gpio: gpio_set_level(215): GPIO output gpio_num error
```
</details>

martinberlin, 2020-12-01:
Good at least now starts to do something. Check on ESP32 manuals or in google to use the right GPIOs. Mind that some are only one direction:

>(3139) CALE: connected to ap SSID:HDWLAN password:thEG26eZ
E (3149) gpio: gpio_set_direction(263): GPIO number error
E (3159) gpio: gpio_set_direction(263): GPIO number error
E (3159) gpio: gpio_set_direction(263): GPIO number error
E (3169) gpio: gpio_set_direction(263): GPIO number error
E (3169) gpio: gpio_set_pull_mode(227): GPIO number error
E (3179) gpio: gpio_set_level(215): GPIO output gpio_num error
E (3189) gpio: gpio_set_level(215): GPIO output gpio_num error
E (3189) gpio: gpio_set_level(215): GPIO output gpio_num error
EpdSPI started at frequency: 4000000

Or simply stick to the ones I sent you, that are proved to be working: https://github.com/martinberlin/cale-idf/blob/master/config-examples/sdkconfig_Wave12I48_PSRAM_active

xeijin, 2020-12-02:
@martinberlin OK I think I found the issue from before and resolved, I can now consistently get it to update with this `sdkconfig`

<details>
<summary>sdkconfig</summary>

```
CONFIG_EINK_SPI_MOSI=14
CONFIG_EINK_SPI_CLK=13
CONFIG_EINK_SPI_CS=-1
CONFIG_EINK_DC=-1
CONFIG_EINK_RST=-1
CONFIG_EINK_BUSY=-1
CONFIG_EINK_SPI_CS2=-1
CONFIG_EINK_SPI_MISO=-1
CONFIG_EINK_SPI_M1_CS=23
CONFIG_EINK_SPI_S1_CS=22
CONFIG_EINK_SPI_M2_CS=26
CONFIG_EINK_SPI_S2_CS=19
CONFIG_EINK_SPI_M1_BUSY=32
CONFIG_EINK_SPI_S1_BUSY=21
CONFIG_EINK_SPI_M2_BUSY=18
CONFIG_EINK_SPI_S2_BUSY=4
CONFIG_EINK_M1S1_DC=25
CONFIG_EINK_M2S2_DC=27
CONFIG_EINK_M1S1_RST=33
CONFIG_EINK_M2S2_RST=5
```

</details>

giving me

<details>
<summary>serial output</summary>

```
export IDF_PATH=/Users/hassan/esp-idf
/Users/hassan/.espressif/python_env/idf4.2_py3.9_env/bin/python /Users/hassan/esp-idf/tools/idf.py -p /dev/cu.usbserial-0200D385 monitor
 hassan@HD-MBP  ~/projects/cale-idf-ws   epaper/22 ●  export IDF_PATH=/Users/hassan/esp-idf
 hassan@HD-MBP  ~/projects/cale-idf-ws   epaper/22 ●  /Users/hassan/.espressif/python_env/idf4.2_py3.9_env/bin/python /Users/hassan/esp-idf/tools/idf.py -p /dev/cu.usbserial-0200D385 monitor
Executing action: monitor
Running idf_monitor in directory /Users/hassan/projects/cale-idf-ws
Executing "/Users/hassan/.espressif/python_env/idf4.2_py3.9_env/bin/python /Users/hassan/esp-idf/tools/idf_monitor.py -p /dev/cu.usbserial-0200D385 -b 115200 --toolchain-prefix xtensa-esp32-elf- /Users/hassan/projects/cale-idf-ws/build/cale-epd.elf -m '/Users/hassan/.espressif/python_env/idf4.2_py3.9_env/bin/python' '/Users/hassan/esp-idf/tools/idf.py' '-p' '/dev/cu.usbserial-0200D385'"...
--- idf_monitor on /dev/cu.usbserial-0200D385 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7084
load:0x40078000,len:13652
load:0x40080400,len:4632
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2-rc-5-g511965b26-dirty 2nd stage bootloader
I (29) boot: compile time 00:00:19
I (29) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[... 58 lines cut ...]

SPI_M1_BUSY:32 <- all set as input,pullup GPIOs
SPI_S1_BUSY:21
SPI_M2_BUSY:18
SPI_S2_BUSY:4

M1S1_DC:25 <- all set as output GPIOs
M2S2_DC:27
M1S1_RST:33
M2S2_RST:5

Available heap after Epd init:4299168
Resolution setting
Sending a buffer[160392] via SPI

Available heap after Epd update: 4300288 bytes
STATS (ms)
879 _wakeUp settings+send Buffer
2702 _powerOn
3581 total time in millis
```

</details>

However, the problem is only the top right of the 4 eink screens updates, and the quality is very poor .. what would be the next step?

martinberlin, 2020-12-03:
I will recheck again the Waveshare commands for Wave12I48RB this afternoon. But to me it sounds like there is a wiring issue, because I had the same, when started doing my hack and in my case only top was updating (S2 & M2). Depending on what's your orientation, I consider USB pointing down, then you are getting updated only M2 (But if it is on other direction can be M1)

[image]

I think must be a wiring issue since this is internally a Master->Slave configuration (We think) that's why it's labelled S & M. 
Also can you try to stick to my configuration that is proved to be working, it should not make any difference but just in case: 

```
CONFIG_EINK_SPI_M1_CS=23
CONFIG_EINK_SPI_S1_CS=22
CONFIG_EINK_SPI_M2_CS=21  --> This is also different
CONFIG_EINK_SPI_S2_CS=19
# ** This you have different 
CONFIG_EINK_SPI_M1_BUSY=32
CONFIG_EINK_SPI_S1_BUSY=26   **
CONFIG_EINK_SPI_M2_BUSY=18
CONFIG_EINK_SPI_S2_BUSY=4
# This is Ok since you have the same
CONFIG_EINK_M1S1_DC=25
CONFIG_EINK_M2S2_DC=27
CONFIG_EINK_M1S1_RST=33
CONFIG_EINK_M2S2_RST=5
```

Will report back after checking my SPI commands with the ones in Waveshare example. Sadly I don't have one of this to test it in my studio, I only have the black & white model.

xeijin, 2020-12-03:
@martinberlin agree it is a wiring issue, just tried with the working config you mentioned and same issue.

Question - are you doing some wiring under the waveshare PCB? Where are the black and red jumper leads in your picture going to?

The GND pins on my ESP32 board all match-up with the GND holes on the waveshare PCB - so I think this is correct

martinberlin, 2020-12-03:
Hi @xeijin 
I added some "Power setting" commands that where missing in the wakeUp() method. Please pull the latest updates, revise your wiring setup, and report back the results. I hope with this we make a step further

martinberlin, 2020-12-03:
>Question - are you doing some wiring under the waveshare PCB? Where are the black and red jumper leads in your picture going to?

No, it's just like I mentioned in the Wiki I'm just re-routing pins 16 & 17 since you cannot use them because of the SPI Ram:

Original WaveI48 Board   | Updated Firmware Pins in ESP32
CONFIG_EINK_SPI_M2_CS=16    21    
CONFIG_EINK_M2S2_DC  =17    27

I'm you sure you read that. So the cables you see are just going from 16 to 21, and from 17 to 27. But that's all.
I think it would be also a great idea to use the same Board I used, a WROVER B, since you can just comb the Waveshare PCB and then you don't need to rewire everything. Only that 2 cables.

xeijin, 2020-12-04:
OK I have soldered headers to a brand new ESP32, here is the wiring:

```
[ESP32] GPIO 21 ---jumper wire--> P16 [WaveShare PCB]
[ESP32] GPIO 27 ---jumper wire---> P17 [WaveShare PCB]
```

I am using the `sdkconfig_Wave12I48_PSRAM_active` with no changes (other than renaming to `sdkconfig`)

still only one corner of the display refreshing, logs:

<details>
<summary>serial output</summary>

```
--- idf_monitor on /dev/cu.usbserial-0200d39b 115200 ---
--- quit: ctrl+] | menu: ctrl+t | help: ctrl+t followed by ctrl+h ---
ets jun  8 2016 00:22:57

rst:0x1 (poweron_reset),boot:0x17 (spi_fast_flash_boot)
configsip: 0, spiwp:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:dio, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7084
load:0x40078000,len:13652
load:0x40080400,len:4632
0x40080400: _init at ??:?

entry 0x400806f4
i (29) boot: esp-idf v4.2-rc-5-g511965b26-dirty 2nd stage bootloader
i (29) boot: compile time 16:30:25
i (30) boot: chip revision: 1
i (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
i (41) boot.esp32: spi speed      : 40mhz
i (46) boot.esp32: spi mode       : dio
i (50) boot.esp32: spi flash size : 4mb
i (55) boot: enabling rng early entropy source...
i (60) boot: partition table:
i (64) boot: ## label            usage          type st offset   length
i (71) boot:  0 nvs              wifi data        01 02 00009000 00006000
i (79) boot:  1 phy_init         rf data          01 01 0000f000 00001000
i (86) boot:  2 factory          factory app      00 00 00010000 00100000
i (94) boot: end of partition table
i (98) boot_comm: chip revision: 1, min. application chip revision: 0
i (105) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x157b8 ( 87992) map
i (151) esp_image: segment 1: paddr=0x000257e0 vaddr=0x3ffb0000 size=0x02328 (  9000) load
i (155) esp_image: segment 2: paddr=0x00027b10 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _windowoverflow4 at /users/xeijin/esp-idf/components/freertos/xtensa/xtensa_vectors.s:1730

i (158) esp_image: segment 3: paddr=0x00027f1c vaddr=0x40080404 size=0x080fc ( 33020) load
i (182) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x1e108 (123144) map
0x400d0020: _stext at ??:?

i (234) esp_image: segment 5: paddr=0x0004e130 vaddr=0x40088500 size=0x0729c ( 29340) load
0x40088500: vringbufferreturnitem at /users/xeijin/esp-idf/components/esp_ringbuf/ringbuf.c:1263 (discriminator 1)

i (257) boot: loaded app from partition at offset 0x10000
i (257) boot: disabling rng early entropy source...
i (258) psram: this chip is esp32-d0wd
i (263) spiram: found 64mbit spi ram device
i (266) spiram: spi ram mode: flash 40m sram 40m
i (272) spiram: psram initialized, cache is in low/high (2-core) mode.
i (279) cpu_start: pro cpu up.
i (283) cpu_start: application information:
i (288) cpu_start: project name:     cale-epd
i (293) cpu_start: app version:      v0.9.8-68-g5552357
i (298) cpu_start: compile time:     dec  4 2020 16:30:11
i (305) cpu_start: elf file sha256:  1391c7a476306a7f...
i (311) cpu_start: esp-idf:          v4.2-rc-5-g511965b26-dirty
i (317) cpu_start: starting app cpu, entry point is 0x40081b78
0x40081b78: call_start_cpu1 at /users/xeijin/esp-idf/components/esp32/cpu_start.c:287

i (0) cpu_start: app cpu up.
i (1209) spiram: spi sram memory test ok
i (1209) heap_init: initializing. ram available for dynamic allocation:
i (1209) heap_init: at 3ffae6e0 len 00001920 (6 kib): dram
i (1215) heap_init: at 3ffd9f18 len 000060e8 (24 kib): dram
i (1222) heap_init: at 3ffe0440 len 00003ae0 (14 kib): d/iram
i (1228) heap_init: at 3ffe4350 len 0001bcb0 (111 kib): d/iram
i (1235) heap_init: at 4008f79c len 00010864 (66 kib): iram
i (1241) cpu_start: pro cpu start user code
i (1246) spiram: adding pool of 4096k of external spi memory to heap allocator
calepd component version 1.0
wave12i48rb() constructor injects io and extends adafruit_gfx(1304,984) pix buffer[160392]

available heap after epd bootstrap:4216508
i (1272) spi_flash: detected chip: generic
i (1276) spi_flash: flash io: dio
i (1280) cpu_start: starting scheduler on pro cpu.
i (0) cpu_start: starting scheduler on app cpu.
i (1290) spiram: reserving pool of 32k of internal memory for dma/internal allocations
calepd version: 1.0
fonts-demo. free heap: 4299972 (after epaper instantiation)
dram     : 4332700
pin setup:
spi_m1_cs:23 <- all set as output gpios
spi_s1_cs:22
spi_m2_cs:21
spi_s2_cs:19

spi_m1_busy:32 <- all set as input,pullup gpios
spi_s1_busy:26
spi_m2_busy:18
spi_s2_busy:4

m1s1_dc:25 <- all set as output gpios
m2s2_dc:27
m1s1_rst:33
m2s2_rst:5

available heap after epd init:4299168
power setting (new)
sending a buffer[160392] via spi

available heap after epd update: 4300288 bytes
stats (ms)
879 _wakeup settings+send buffer
4712 _poweron
5591 total time in millis
```
</details>

martinberlin, 2020-12-04:
If you pull last updates, and still does the same, it must be some Wiring issue. But still to clear any doubt, can you default to the Waveshare wiring and upload their example?
I want to make sure it’s not something from the epaper itself...
Serial looks just fine. What I don’t understand is why only one epaper updates.
The beginning should be to start first with the default wiring and Waveshare demo. If you use Platformio I left the default here:
https://github.com/martinberlin/esp32-epaper-12.48-waveshare

Other than that can you get an ESP32 WRover B that has same pin out and then you place it just there? That saves a lot of time and you don’t have to rewire everything.
Check also my class and the Waveshare (B) example just to see if I missed something.

xeijin, 2020-12-04:
@martinberlin I am using a board with the WROVER-B chip, our board manufactuers are different - I think your version was a WeMos D32 but that doesn't seem to be available on sale in the UK anymore. Only difference are (1) my board comes with a battery attached (2) pin headers are not pre-soldered on my board - pin layout matches WaveShare PCB exactly.

I am [using this model instead](https://www.amazon.co.uk/ICQUANZX-Bluetooth-Battery-ESP32-WROVER-B-Development/dp/B07VQJ1RH2/ref=pd_sbs_200_1/262-0163938-5202069?_encoding=UTF8&pd_rd_i=B07VQJ1RH2&pd_rd_r=159869bc-04d2-42e8-b8e7-938176466410&pd_rd_w=DVdhv&pd_rd_wg=wBfeU&pf_rd_p=2304238d-df78-4b25-a9a0-b27dc7bd722e&pf_rd_r=8SWY3JN2K2P6TQJPVX2Y&psc=1&refRID=8SWY3JN2K2P6TQJPVX2Y) as you can see from the images, pin layout is the same.

I am using the latest commit from the `epaper/22` branch - will try your original repo and revert back.

xeijin, 2020-12-04:
@martinberlin I have also tried martinberlin/esp32-epaper-12.48-waveshare:

If choosing '12.48inch e-paper b' writing a string in red and clicking 'upload image' it prints once in serial output '12.48 inch e-Paper b` but nothing happens on the display

```
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- Miniterm on /dev/cu.usbserial-0200D39B  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
.
WiFi connected
Server started, ip: 10.10.100.186
epaper.local mDns started
HTTP server started

Ok!

12.48inch e-Paper b
```

If I choose the black and white version, draw the same string in black, I get more console output, but no change to display and no further console output

```
WiFi connected
Server started, ip: 10.10.100.186
epaper.local mDns started
HTTP server started

Ok!

12.48inch e-Paper
M1 Busying
M1 Busy free
EPD_12in48_Clear start ...
over ...
```

martinberlin, 2020-12-04:
But wait, this version of Waveshare opens a server in the ESP32. And you have to enter to that IP address and then after drawing something you can send it to the epaper. 
Check : https://www.waveshare.com/wiki/12.48inch_e-Paper_Module "WiFi example"

xeijin, 2020-12-04:
> But wait, this version of Waveshare opens a server in the ESP32. And you have to enter to that IP address and then after drawing something you can send it to the epaper.
> Check : https://www.waveshare.com/wiki/12.48inch_e-Paper_Module "WiFi example"

Yes I was doing exactly this - the serial outputs are based on writing a string using the web client then clicking the upload button.

xeijin, 2020-12-04:
@martinberlin is it possible to find out which pins these errors relate to? Maybe some deeper debug level or something? I noticed these errors when switching to the `cale.cpp` component.

```
E (3150) gpio: gpio_set_direction(263): GPIO number error
E (3160) gpio: gpio_set_direction(263): GPIO number error
E (3160) gpio: gpio_set_direction(263): GPIO number error
E (3170) gpio: gpio_set_direction(263): GPIO number error
E (3170) gpio: gpio_set_pull_mode(227): GPIO number error
E (3180) gpio: gpio_set_level(215): GPIO output gpio_num error
E (3190) gpio: gpio_set_level(215): GPIO output gpio_num error
E (3190) gpio: gpio_set_level(215): GPIO output gpio_num error
```

xeijin, 2020-12-04:
> after reseating all the pins in the dupont jumper wires, and re-soldering all the joints, then testing everything with a multimeter ... I am getting a slight improvement (now get a picture in 2 screens ... each opposite corner, and red colour is appearing, but still no image on the remaining 2 screens and quality remains poor.
> 
> I guess it must be wiring, perhaps my poor soldering skills ... but not sure what else to try?
> 
> PS: I disconnected the EPS32 and tried using a raspberry pi in the waveshare PCB and the display worked fine first time ... so has to be either my wiring/soldering, the library or maybe a problem with the ESP32 socket?
> 
> [image]
> [image]
> [image]
> [image]

ok I reverted back to `wave12i48.h` (black only version) library and now I get an image across the two upper screens...same confit as before, only re-flashed haven’t touched any wiring

[image]

martinberlin, 2020-12-05:
Ok that's a progress.
The bottom part is missing. With the wave12i48BR Black/red version is strange, because you are getting only the M2 and M1 panels working. Will check again to see if I'm missing something on the BR version but I'm quite sure I copied all the initialization commands. 
Since the M1 / S1 down panels are not working I would specially check pins on that ones:
#define EPD_M1S1_DC_PIN  25  and so on. 

Did you also make sure that all 3 GND pins of the Waveshare PCB are connected to ground?
Apropo the soldiering, I also have bad soldiering skills and not a good precision solder, that's why I strongly recommend trying to do this with an ESP32 WROVER B:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/modules-and-boards.html

If you use one of this, you only have to re-route this two pins, what you show is OK and is the way to do it. But the benefit is that then this connect directly to the Waveshare PCB, so you avoid having this wiring/connectivity issues.
Other than that, if it works better with my only black class, there must be something wrong with the other. I will double check.
You live in China?
I would like to try one of this to correct my class but I really don't want to spend 180 u$ for something I don't need just to make the class. But if you have one extra someday, you can send it to me, I will make the class and then send it back to you with an additional working and tested ESP32 WROVER mounted. If this sounds like a deal, I will pay the sending costs back and forth to you.

xeijin, 2020-12-05:
@martinberlin made some progress - 

I can confirm the wiring is fine - I managed to get the display working 100% with the WaveShare demo code:

- I cloned https://github.com/waveshare/12.48inch-e-paper
- imported folder `esp32/wifi` using PlatformIO as an Arduino project (chose board: `lolin_d32`, same as in your project)
- changed pins in `src/DEV_Config.h` (`EPD_M2_CS_PIN 21` and `EPD_M2S2_DC_PIN  27`)
- updated WiFi ssid/password in `src/WebApp.cpp`
- flashed to ESP32

I was then able to display the following - note all 4 areas of the screen are used (and quality is no longer poor)

[image]

I tried flashing back to the `cale-idf` firmware immediately afterwards (without moving the device at all) - result was the same as before. I even shook the cables a bit then reflashed the waveshare demo code and it worked - think this means wiring is fine and problem is with the class?

Will keep investigating.

PS - this board is WROVER-B, see this image of the chip:

[image]

This model actually has 8MB of PSRAM (though I believe only 4MB is addressable by ESP32 anyway)

martinberlin, 2020-12-06:
Great. Can you please double check the SPI initialization commands in the class against the Waveshare ones?
Note that in update() method I'm only sending the black right now. The idea was to make it work first and then check the color.
If I see that correctly [EPD_12in48b.cpp](https://github.com/waveshare/12.48inch-e-paper/blob/master/esp32/wifi/EPD_12in48b.cpp) is the class for the Red/Black version.

Good to know that the Eink is working that is the first step. Let's try to make the class work also. Cannot be so complicated. We only to send the right commands. And the black only version is working, so we can use both the EPD_12in48b.cpp from Waveshare and my own black version that are working as a base for this new class.

>ok I reverted back to wave12i48.h (black only version) library and now I get an image across the two upper screens...same confit as before, only re-flashed haven’t touched any wiring

One thing, now that you have the wiring correct and working, can you also try flashing something with the wave12i48.h (black only version)
But don't use Cale.cpp use some demo-font or anything that sends pixels directly from C++ using Adafruit GFX. Because the cale example is downloading a bitmap that is not high quality. That's probably why you get this results.

martinberlin, 2020-12-11:
Hello @xeijin,
can you give me a bit more feedback about this?
Another user wrote me that is trying to do the same. If I see that there is more interest about this I will probably sell my black/white version and buy the red/black. But it will take some time.

xeijin, 2020-12-12:
@martinberlin sorry been a busy week at work, will be trying your latest suggestions tomorrow and reporting back

xeijin, 2020-12-12:
@martinberlin I tried taking a look at the code but I think this is a bit beyond my abilities ... are these the lines I should be trying to change in `wave12i48BR.cpp`? https://github.com/martinberlin/cale-idf/blob/epaper/22/components/CalEPD/models/color/wave12i48BR.cpp#L124-L259

martinberlin, 2020-12-12:
Hello!
Yes exactly. You can also comment the setLut() at the end of wakeUp() method.
Actually driving the epaper is not hard. This one is specially complex since it has 4 epapers in one. But essentially is resetting it (RST) pin. Then sending some initialization commands (one of them is power on) then comes the update, where it receives a command that signals that we will send the pixel buffer and then Refresh. That’s how most of them work.
But you are getting good in IDF not all people arrive to compile this. So don’t loose the hope. In the worst case I will sell mine and get a red/black one

xeijin, 2020-12-18:
@martinberlin thanks for the pointers! I managed to find the issue, quality issues have gone and display is refreshing properly now:

[image]

I guess only remaining issue is that it is showing black as red, presumably because color support is not fully implemented yet in this class?

martinberlin, 2020-12-18:
Great!  Very happy that it works. Now you can draw things from C++

Yes check in the Waveshare class that there are 2 commands one is to refresh black and another to do it with red.
What was the issue? Would be interesting to document it better.

The next step is to add the red buffer. However needs to be done with care since should be placed in PSRAM otherwise there won’t be Ram available since it’s a big buffer.
Will check it today’s afternoon and update this.

martinberlin, 2020-12-18:
Thanks for the fix of my SPI commands. According to [Waveshare documentation for this epaper](https://www.waveshare.com/wiki/12.48inch_e-Paper_Module_(B)):

>Registers 0x13 and 0x10 are used to control Black-white image data and Red image data transmitting

But real commands guide PDF is here: https://www.e-paper-display.com/GDEW1248Z95%20V1.1%20Specification11a8.pdf?method=picker&flag=all&id=de105ef3-31d1-4ef1-b6a8-9b63cd4a2652&fileId=679&v=0.zip

And there it specifies that:
 0x10 is DTM1, White/Black Data
 0x13 is DTM2, White/Red Data

Now I switched to 0x10 in the **epaper/22** branch and it should update only black. If that works we keep on implementing the Red buffer.

martinberlin, 2020-12-19:
Xeixin, when you have time I left you 2 tests prepared:

1.  **epaper/22** branch. Supports only one color, has already your FIX of the initial SPI commands

2. **epaper/22-ext-ram** branch. I added the Black / Red support but is very possible does not work. I'm assigning both buffers directly in PSRAM on the class. Also updated the method drawPixel so it supports red and the update so it sends the buffer

Can you please give me feedback when you find some minutes to test it?

xeijin, 2020-12-20:
@martinberlin the screen does not seem to be clearing correctly now, with both of the branches you provided, and actually even my original build - but I think this is my mistake (because original waveshare demo is still working, so not a hardware problem) - probably I made some change without committing (beginners mistake), let me retrace what I did - I will come back to you and confirm.

martinberlin, 2020-12-20:
Thanks for testing!
I found the bug. We should do a fillscreen WHITE now that I corrected it. Please pull again **epaper/22-ext-ram** branch and flash it. Let's see if we can make it work!

xeijin, 2020-12-21:
@martinberlin getting there! Definitely an improvement though the colours are a bit faded/poor quality

this was with *epd/22-ext-ram*, fresh checkout & pull 

[image]

I tried *epd/22* again (separate clone, checkout & pull) but strangely I am getting similar results to the other branch except the text is white ... doesn’t sound right to me, have probably made a mistake somewhere

[image]

I am off work tomorrow so going to double check my sdkconfig and I will also compare to my previous class changes to see if I did anything different.

martinberlin, 2020-12-21:
Ok I think that in epaper/22 you are getting the same red since we are not touching the red channel at all. The white is quite strange, cool effect though never seen something alike.

Let’s focus our efforts in the Branch with the red channel.

Try to draw something in red too, for example: display.fillCircle(x,y,radius)
Check not sure if parameters are right. I will check on my side to see if the red channel is implemented correctly. Good thing to see that the arrays savedin the external RAM seem to work as expected!

xeijin, 2020-12-21:
@martinberlin **it works!** 🥳 🎉 it looks like the [old issue with spi power commands](https://github.com/martinberlin/cale-idf/pull/25/commits/c508b2526c6046f5f39745ea0b4a330e2cddaae4#diff-a7d15c3905f2873e7ddc45dfbee6eef9e5331708f6f270b5a1af7fe3616c151c) had snuck back in.

I'm now getting a uniform, deep red colour:

[image]

remaining issues:

- `EPD_BLACK` is painting red and`EPD_RED` is painting black - I am guessing I can fix this by playing around with the inversion settings
- `EPD_WHITE` does not seem to paint at all (or maybe it's painting red also?)
- when the screen is refreshing it makes a slightly higher pitched sound (not very loud, but noticeable in a quiet room) I dont think the waveshare firmware does this but will double check

martinberlin, 2020-12-21:
Awesome. Yes there is definitively an issue with White.
Looking at Waveshare clear function, 8 pix of white are 0x00, and 8 pix ink in the black channel are 0xFF which makes sense. Now there is an inversion for the red but maybe I got it back. Let me check it again, but we are very close!

xeijin, 2020-12-21:
@martinberlin can you confirm which file? I am trying this in `demo-fonts.cpp` let me know if not correct.

- [ ] `FAIL` **fillScreen(EPD_WHITE); return; // Just stop there and check that the screen is white, nothing printed.**
```cpp
   display.setRotation(2); // 0 - 12.48 w/USB pointing down
   display.fillScreen(EPD_WHITE);
   display.update();
   return;
```
I had to add `display.update();`, screen clears to **black** instead of white

logs

```
Sending BLACK buffer[160392] via SPI
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
Sending RED buffer[160392] via SPI
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
Available heap after Epd update: 4139920 bytes
STATS (ms)
1450 _wakeUp settings+send Buffer
4711 _powerOn
6162 total time in millis
```
    - no effect, display does not change at all (still showing the previous image)

- [ ] `FAIL` **Remove stops and let the whole example run**
  - I am getting a black background, one red circle and one black circle

- [X] `SUCCESS` **fillScreen(EPD_BLACK); // And invert setTextColor(EPD_WHITE)+fillCircle(x,y,rad,EPD_WHITE). Let the whole example run again.**

[image]

xeijin, 2020-12-21:
@martinberlin also before trying your instructions (using code in #25), I changed only the `8PIX` hex values in `wave12i48BR.h` and got the same result...

[image]

martinberlin, 2020-12-21:
> I am trying this in demo-fonts.cpp let me know if not correct.

sure that is the right one.

Oh that’s strange. Then the white/black is inverted right? 
Awesome, we are almost there!

A small correction more. Give me some minutes

xeijin, 2020-12-21:
@martinberlin  I have managed to fix it with some trial-and-error, all colours, whether using `fillscreen()` or the adafruit library are showing as expected. Please see my PR, #25 which should be the complete set of changes to get it working.

[image]

martinberlin, 2020-12-21:
Merged in master. Great team work!
I guess we can close this now, but just for the sake of a last test, feel free to pull cale-idf master branch and compile the last run.

Good news: Now that we know how to store the pixel buffer arrays in the PSRAM of the ESP32 we could even drive much bigger epapers.

martinberlin, 2020-12-21:
Looking forward that you can create an account in CALE and render some awesome Black/Red Bitmap. Just tell me if you need any assistance. The CPP to compile should be: 

**main/cale.cpp**

And you should specify in menuconfig the URL that you get in Cale->Render image and setup your WiFi credentials.  Mind that in Render image you can also take profit of Image dithering so you get simulated grays.
I recommend trying the image Gallery, where you can upload multiple images, and on each request you get the next one. So you can make kind of a Photo-frame slider.

xeijin, 2020-12-21:
@martinberlin I just used [the URL embedded in your PSRAM sdkconfig](http://img.cale.es/bmp/fasani/5ef2ee980a4ec) and got a bit of a surprise haha 🤣 

[image]

it definitely works though, amazing!

it did not work the first try, had to make a few modifications (included below for anyone who comes across this later):

`main/cale.cpp`:

add the right include

```cpp
// color version
#include "wave12i48BR.h" // Only to use with Edp4Spi IO
Epd4Spi io;
Wave12I48RB display(io);
```

`main/CMakeLists.txt`:

- comment `SRCS "demos/demo-fonts.cpp"` (otherwise it will complain there is not enough space)
- uncomment `SRCS "cale.cpp"`

`sdkconfig`

- after copying over from `sdkconfig_Wave12l48_PSRAM_active` I had to make the following changes:
  - update Wifi SSID & Password
  - `CONFIG_DISPLAY_ROTATION=2` otherwise display rotation was incorrect

cale.es looks like it has alot of features, I have just signed-up but will need a bit more time to play around with it! Thanks again!

closed 2020-12-21 by xeijin

martinberlin, 2020-12-21:
Great! Very happy that it does work.
Perfect, you did all the right steps. That is a gallery that I prepared from a quite famous german photographer called Helmut Newton

martinberlin, 2020-12-22:
Hi @xeijin,
I have an additional request whenever you find some time. Can you print some nice photo / drawing on this big epaper?
Then you can just take a nice quality picture (If possible with natural light, or good light)
So I can use it to illustrate this [Model Wiki page](https://github.com/martinberlin/cale-idf/wiki/Model-color-wave12i48BR.h). Thanks a lot!

xeijin, 2021-01-03:
@martinberlin hey just saw this, away for a couple of but will definitely do this when I get back

martinberlin, 2021-01-03:
No problem, is not urgent, whenever you might find some time. Thank you very much!