martinberlin/cale-idf #28: Issue 17 touch. Does it work with bigger displays?

DominicD, 2021-01-24:
Hi I am looking for exactly this. However I want to use a bigger display. Would this also work with one of the bigger displays and touch panels?

martinberlin, 2021-01-29:
I'm not sure since I never tried. But the manufacturer is also Focal Tech so it should work provided I also integrate it in the 4.2 Class. 
Just tell me if you want to use Cale-idf or you are just interested in the touch. You can also reach me by email at martin at luckycloud.de

martinberlin, 2021-03-02:
Any feedback here?
Will you get one of this 4.2?   I highly recommend the new T5S from LilyGo that comes also with touch (See Lilygo official store in Aliexpress)
Much more resolution and much faster.

closed 2021-03-09 by martinberlin