martinberlin/cale-idf #29: Add support for 5.65 inch 7 color epaper from Waveshare

martinberlin, 2021-01-29:
This epaper with unknown controller since Waveshare thinks it's such a secret that cannot share what it is, even it is slow as a cow and pretty useless unless you display some artwork on it... it's coming finally to Cale-idf.
[image]

I will take next two weekends to implement it. I really do not want to do it the GXepd2 way since I don't like how the whole class it's made, so I will just use the Waveshare example and do my own take with it.

martinberlin, 2021-01-30:
[image]
Implemented

martinberlin, 2021-02-08:
Implemented and merged in master. Leaving this open since it comes soon a 4.01" 7 colors epaper

martinberlin, 2021-02-21:
4.01 new class added: gdew0583z21.cpp
[image]

ilachEU, 2021-03-12:
In fact, these e-papers can display colors up to 8.
https://youtu.be/YawP9RjPcJA (2:27)

martinberlin, 2021-03-12:
That’s interesting a pity is not revealed In the video how to get the pink color. Will have to ask him ;)

ilachEU, 2021-03-12:
He has 2 long streams stored on his channels, he explains it there.  It's actually the last color 0x7.

martinberlin, 2021-03-12:
Ok cool let's give it a try. According to the [docs it has only from 0x0](https://www.waveshare.com/wiki/4.01inch_e-Paper_HAT_(F)) to 0x6 but it be cool to see a new color to appear. Let's see if I can add it at least to use in GFX

- [ ] Add the wild pink 0x07 color

martinberlin, 2021-03-12:
@ilachEU new color added. Constant is EPD_PURPLE 

```C
//                                         565         888
#define EPD_PURPLE    0x41E8 //  67,  61,  68 
```

ilachEU, 2021-03-12:
Thumb up !  Have you tried it too?  It works ?

martinberlin, 2021-03-12:
Yes sure it does (top left)
[image]

martinberlin, 2021-03-14:
8th color added thanks @ilachEU for the feedback
Documentation added for both epapers:

- [wave4i7color 4.01 inches](https://github.com/martinberlin/cale-idf/wiki/Model-color-wave4i7color.h)
- [wave5i7color 5.65 inches](https://github.com/martinberlin/cale-idf/wiki/Model-color-wave5i7color.h)

closed 2021-03-14 by martinberlin