martinberlin/cale-idf #31: Model gdep015OC1.h does not work

DominicD, 2021-02-07:
Hi I have now bought this mode l from Waveshare.
I try to get the demos/demo-epaper.cpp running, however it does not display anything.
I have checked all the oins and the sdkconfig and they should be correct.

Do you have any tips on what I am doing wrong?

DominicD, 2021-02-07:
I think I found the problem. Seems there is a new Version from waveshare which uses different Display. 
I tried with GxEPD2 and the model GDEH0154D67 works.

DominicD, 2021-02-07:
Maybe you can tell me how you went about implementing it. E.g. where you got the lut and commands and other information from. I will try to integrate it

martinberlin, 2021-02-07:
Ah ok I didn't know that. Can you send me the link to the GDEH0154D67 class in GxEPD2?
I based this project in GxEPD luts / initialization commands and the Waveshare / Goodisplay demos itself. So sure if you want just copy existing model and try to make the necessary modifications.

Basically it's initialization commands and the right command before sending the pixel buffer + refresh. Just tell me if you need help.

DominicD, 2021-02-07:
If I make a pull request should it be against this repor or https://github.com/martinberlin/CalEPD?

martinberlin, 2021-02-08:
Against this one is ok!  Then I test and review updates, like once a month, I make a new CalEPD release.
So the things are developed and tested here, then I export the updates to the independent components. No idea if it's the best methodology but it worked so far. Thanks a lot for your collaboration

martinberlin, 2021-02-09:
Can you upload here a photo of this epaper so I add it also in the Wiki?
I will review your pull request and merge it today. Thanks!

DominicD, 2021-02-09:
Sure. I tried the demo-epaper.cpp and the demo-sleep-clock.cpp. Have not tried yet partial update. 
I will post pictures as soon as I can

martinberlin, 2021-02-09:
No problem. Great that it works, I will merge it in master, and then clean up some deprecated methods.

martinberlin, 2021-02-09:
Leaving this open @DominicD so you can upload a picture in the ticket whenever you find some time.  Thanks a lof for your contribution

DominicD, 2021-02-09:
[image]

Partial refresh seems to be buggy:
[image]

martinberlin, 2021-02-09:
Ups partial refresh needs a fix right?
Thanks for the pictures, can you try partial refresh with gxEPD2 and see if it works?

if it does I will compare he two classes and see if I can find something.

DominicD, 2021-02-10:
[image]

Partial update works fine with GxEPD2

martinberlin, 2021-02-10:
Thanks for the update. Tried to implement it but I cannot test:

[refactor/oop bf2caa1] #31 Implement partial update the gxepd2 way. Just call updateWindow(x,y,width,height)

Just checkout refactor/oop and try to run first demo-epaper full test. And then add somewhere an updateWindow(x,y,w,h) and try partial.

DominicD, 2021-02-10:
It is still buggy

DominicD, 2021-02-10:
If you want we could test it together to find out what the problem is.

martinberlin, 2021-02-10:
Sure this Friday or in the weekend I‘ve time. But buggy in the same way as before?
Does full update still work?
I tried to repeat same commands as gxEPD2 but is hard to get it the first time without having the Epaper. But for sure let’s find some time and do some screen sharing. I will check anyways and see if I didn’t missed anything.

DominicD, 2021-02-11:
The full update still works. 
In your wiki you mention that the old model gdep015OC1.h is also buggy with partial refresh do you still have that one? Maybe it helps you if you fix that one.

Sure we can try to work on it together. Do you have discord or anything else? Maybe you can share your contact details via E-Mail?

martinberlin, 2021-02-11:
Sure you can email me at Martin at luckycloud.de
I also use discord but I don’t remember my user at the moment. Let’s compare together if I’m missing something, following gxEPD2 way. There must be something that is different!

martinberlin, 2021-02-14:
Great find @DominicD 
Now partial update works as expected (No more white points) it seems it was the power on mode that you found. Thanks.
Now I will cleanup and try to optimize SPI

martinberlin, 2021-02-14:
Nice teamwork, will wait till the end of next week before closing, feel free to suggest any optimization if you feel like @DominicD 
Full update working, partial refresh working. Wiki entry:  https://github.com/martinberlin/cale-idf/wiki/Model-gdeh0154d67.h

martinberlin, 2021-02-16:
New epaper watch in the making with partial refresh
[image]

martinberlin, 2021-03-02:
Closing down here. @DominicD thanks for your collaboration

PR #32 (2021-02-09): Added gdeh0154d67. 
Closes #31
This is the display used in Waveshare 1.54 inch Rev.2.1 Model https://www.waveshare.com/product/1.54inch-e-Paper-Module.htm
closed 2021-03-02 by martinberlin