martinberlin/cale-idf #33: Error in handling of padding in cale.cpp

DominicD, 2021-02-10:
I adapted your code for my need to print a bitmap from memory. What I found is you show the correct image, however you are drawing too many pixels. 
E.g. when there is a width of 200 pixels, your calculations are giving rowSize = 28 = 224 pixels. Which is fine, however what you are doing is drawing pixels line by line from 0 to 224 which is too much. You should only draw until 200. 
You can see this in your debug output: 
`
printf("Total drawPixel calls: %d\noutX: %d outY: %d\n", totalDrawPixels, drawX, drawY);
`
Which will for a 200x200 pic emit 44800 (200x224) pixels.

I have it working for my needs however I am not sure how to fix it for your code.

martinberlin, 2021-02-13:
Good find, can you suggest a correction with a pull request and I test if it works?
If so I can check it out with different sizes and merge it once it's tested

martinberlin, 2021-03-02:
@DominicD  what can we do here?
Can you suggest a correction and we test together?

So far I didn't have any problem with loading Bitmap images that's why I didn't noticed this. But any optimzation is welcome :)

martinberlin, 2021-03-11:
Closing here. Please reopen and we propose a solution in case this is affecting some Firmware

closed 2021-03-11 by martinberlin