martinberlin/cale-idf #35: Support for Waveshare 7.5 in HD?

KonstantinSchubert, 2021-03-07:
Hi Martin,

I have used CalEPD, in particular[ this class,](https://github.com/martinberlin/cale-idf/blob/master/components/CalEPD/models/gdew075T7.cpp) as a reference when implementing my code to drive my [Waveshare 7.5 in V2](https://www.waveshare.com/product/displays/e-paper/epaper-1/7.5inch-e-paper.htm) display.

There is now a new display, the  Waveshare 7.5 in **HD** display, and it seems to have significantly different driver needs.

I have been looking in your code to find a class that supports this new HD display, but I only found [this one](https://github.com/martinberlin/cale-idf/blob/master/components/CalEPD/models/gdew075T8.cpp): That's not the right one, right? What display is that one for?

martinberlin, 2021-03-08:
I never had this Epaper (880×528) so never made this class. If you want you can make a copy of the T7, update the SPI wakeUp commands. Then just send it to me or make a pull request 

This can be the only way to do it remotely since I'm not planning to buy one of this soon (unless some client wants it) looks like a nice epaper.
No idea if the buffer fits an ESP32 without PSRam that you need to try yourself.

KonstantinSchubert, 2021-03-08:
I've been trying to do what you suggest, but the device doesn't ever get "BUSY" after sending the data  - and then it doesn't update.

One interesting difference between the V2 and the HD display seems to be that the indicator for BUSY seems to have been changed  from LOW to HIGH. 

I wonder if that has any impact on the GPIO settings, in particular this line 
```
ESP_ERROR_CHECK(gpio_set_pull_mode((gpio_num_t)CONFIG_EINK_BUSY, GPIO_PULLDOWN_ONLY));
``` 
in `epdspi.cpp`...

martinberlin, 2021-03-08:
Check well the tech specs PDF. Looking at the Waveshare example class busy is high while is busy and LOW when it finished. Double check the wiring and that the wakeUp commands are exactly like the given ones.

If you cannot get it working just come over one afternoon with the Epaper and we make it work.

KonstantinSchubert, 2021-03-08:
> Bad wiring (power ok? Grounded?) are SPI exactly as defined and in GPIOs that support input/output check config-examples

I am using the [same driver board ](https://www.waveshare.com/product/displays/e-paper/driver-boards/e-paper-esp32-driver-board.htm)both for the V2 display and the HD display. So the GIPOs shouldn't change, right?

martinberlin, 2021-03-09:
No. Then take a close look at initialization commands, and then try it with Waveshare example, see if it works with their demo.

KonstantinSchubert, 2021-03-10:
Awesome :+1:

closed 2021-03-12 by martinberlin