martinberlin/cale-idf #37: Can I use cale-idf in Arduino

cultur98, 2021-03-14:
Hi Martin,
very nice work!
It might be a stupid question but I don't
have issues making a fool of myself. .-)
Can I use the library in Arduino?

martinberlin, 2021-03-14:
Short answer: I've used specific ESP-IDF things regarding SPI Communication (Check for ex: epdspi.cpp )
So of course it could be ported. But is not me who is going to take this task. Primarily since I use arduino-esp32 only to test quick things but most thing this days I build with ESP-IDF.

Why don't you use GxEPD or GxEPD2 that work good and they where made for the arduino framework?
Or is that you want to use Cale-idf for some particular reason?
SPI works faster in this component. But is only because I'm sending full X lines in each transaction. Other than that you could do the same, except of the touch support, in the GxEPD* libraries

cultur98, 2021-03-14:
Hi Martin, thanks for you quick response.
I am using the EPD47 from LilyGo.
And cale-idf seems to be the only library to 
provide screen rotation for it.

martinberlin, 2021-03-14:
Oh I understand that’s a pity. What you can do is fork Lilygo, add rotation based on my class here:
https://github.com/martinberlin/cale-idf/tree/master/components/CalEPD/models/parallel

Basically add a wrapper before write pixel that will add software rotation before drawing it. It’s a pity Lilygo did not add them themselves...
But is not that complicated. From the top of my head: 
Rotation can be a global p: uint8 rotation = 0;

then a function setRotation can update this value from 0 to max 3.
And then the function drawPixel will take this in account and basically recalculate X,Y based on the rotation integer. So instead of calling:

epd_draw_pixel(x, y, color, framebuffer);

you will call your own drawPixel that will take in account rotation just like I’m doing in Ed047TC1::drawPixel(int16_t x, int16_t y, uint16_t color)

that’s mostly it. And then if you get it working just make a pull request and we make rotation work also for those using arduino-ESP32 in this epaper.

martinberlin, 2021-03-15:
@cultur98  tell me what are your plans and if I can help somehow. This will remain open till Friday 19.03 waiting for your feedback

cultur98, 2021-03-16:
@martinberlin 
You can close the case. The device uses a framebuffer. It should be easy to rotate the pixels on demand.
However I use it in "landscape" mode right now. When I am ready for pixel work I will make an approach
and keep you updated!

martinberlin, 2021-03-16:
Alright perfect. Otherwise just suggest in the rotate issue from Lilygo repository (Maybe just copy my proposal)
And when I find the time will just fork it and add it, so other people can use it too, but it would be nice it Lilygo does it. 
I don't mind but I really prefer that they implement it since maybe they want to do it differently. Anyways if they say yes, do it, since they have no pland to implement it I can step in. Keep healthy! 
And try ESP-IDF, you won't regret, it's a very nice Framework to work with.

closed 2021-03-16 by martinberlin

martinberlin, 2021-03-21:
Hi @cultur98 please check my update in Lilygo repository for the EPD47 parallel epaper
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/9#issuecomment-803549946