martinberlin/cale-idf #40: Forking Policy

jockm, 2021-05-13:
I disagree with your fork policy.  First off central control it is somewhat antithetical to the idea of decentralized development.  People fork for other reasons than you mention. For example I often fork vital libraries so I can control when my project decides to update.  I can't always rely on projects having good tagging practices to fix releases.

I have had too many builds break because of this kind of issue to simply trust the maintainers, and I don't think that is an unreasonable assumption.  Also I have had projects disappear on me, either moved to another host without warning or simply deleted for whatever reason.

My point is there are reasons why people fork that are beyond simple "bookmarking", and I think it would be better if you didn't use such perscriptive language for something outside of your control.

You are, of course, free to disagree...

jockm, 2021-05-13:
Also if you want me to only fork components, wouldn't it be better to make the components submodules to make forking just that easier?

jockm, 2021-05-13:
Finally if you say it is better to fork the component (In this case CalEPD), why does the comonent have the same Fork Policy?  It fact the fork policy doesn't even have the comment about it being better to fork this instead of cale-idf

martinberlin, 2021-05-14:
It’s ok to disagree. No problem. My intention is not to have a repository with a lot of forks that are just old copies.
This project is not going to disappear and if I decide to leave it, probably someone else will take the lead. About forking the other components for example if you need just the Epaper part, yes it’s a better idea, since cale-IDF is only where main development takes place, then I make new releases of the components (CalEPD, touch)
So those are clean components, without all examples and other things you won’t need for a real project.

closed 2021-05-14 by martinberlin

jockm, 2021-05-14:
But your submodules have the same forking policy without mentioning it is more acceptable to fork them.

You can't promise your repository won't go away, you can just promise you have no intention now, and it is a bit demeaning to tell people their reasons for forking aren't valid, which is exactly what you are doing.  It also seems to miss the point of your license.

martinberlin, 2021-05-14:
Ok will review that. But yes I’m promising that non of my repositories where I put much effort and people use are not going away. That’s not a valid reason to fork them.
Why such an obsession to have a copy? I really don’t understand this.

martinberlin, 2021-05-14:
Added in CalEPD and touch:

>This repository is not going to dissapear or be deleted by any means so you can use it with confidence.
Forking this in order to make updates that fit your particular project is alright, as long as you update the readme and state what are the intentions to do so.

So I hope that makes things more understandable and you can go ahead if you need to fork that repositories to adapt them to some project.

jockm, 2021-05-14:
@martinberlin 

> Why such an obsession to have a copy? I really don’t understand this.

1. Experience.  Project go away, radically change, move repositories, etc.  And after nearly 4 decades of being an engineer I have seen it happen enough times not to rely on good intentions.  Also services disappear all the time.  We like to think because github is owned by MS that means it is stable, but enough valued services have gone away over history to make one careful about things that are mission critical
2. It is a common practice.  Every large shop I know forks all the repositories they use.  It gives them release stability, and it gives them the option to modify and push if they choose
3. This is open source, forking for whatever arbitrary purpose is a part of the ethos.  If for whatever reason you want to restrict modifications then use a CC*-ND license and have people get permission, or modify your Apache-2.0 to actually have some weight to what you are saying
4. Your tone: If you expressed it as "We would prefer it if you only forked for these reasons..." then I wouldn't have bothered, but you are ordering people into prescriptive behaviors, while you have a deeply permissive license
5. I don't understand the motivation.  Can you point to a true harm of having many forks, aside from your preference?

martinberlin, 2021-05-14:
I simply don’t like it. I don’t like when I want to develop something, if I have to browse the forks, then there are 98% old copies, where no developer even though of changing the title or readme so I know if it had some intention.
I understand your points but this is how I like it and I’m not going to change that. Forking needs  to have an intention. I simply don’t care what the large shops do or if it’s a common bad practice. I simply won’t support it and if a developer does it without respecting my policy I can simply bock him, thus won’t be possible to interact with my repositories anymore.
Thank you for your understanding and the conversation in this issue.