martinberlin/cale-idf #52: Uncommented parallel boards config prevents building for ESP32C3 devices

gdesoto, 2021-12-13:
**Describe the bug**
Using an ESP32-C3 (ESP32-C3-DevKitM-1 dev board) on a fresh copy of the code, the build fails to compile with the following error.

```
In file included from ../components/CalEPD/models/parallel/ED047TC1.cpp:1:
../components/CalEPD/include/parallel/ED047TC1.h:14:10: fatal error: epd_driver.h: No such file or directory
 #include "epd_driver.h"
```

**To Reproduce**
Write meaningful steps to reproduce the behavior:
1. Cloned the repo
2. Uncommented `SRCS "demos/demo-epaper.cpp"` in `main/CMakeLists.txt`.
3. In my case, I set display type to Gdeh042Z21 (so I also had to comment out the display.updateWindow line in demo-epaper but that is unrelated to the core of the issue)
4. Compile the SRC file demos/demo-epaper.cpp for target ESP32C3
5. Fails to build with the error above.

**Expected behavior**
Build Successfully

**Desktop (please complete the following information):**
 - OS: Windows 10

**Additional context**
I was able to resolve the issue by going into the `components/CalEPD/CMakeLists.txt` and commenting out the following lines:
```
#"epdParallel.cpp"
#"models/parallel/ED047TC1.cpp"
```
Interestingly, the comment just above that line says to uncomment those lines for parallel displays, but they were already uncommented. So I'm not sure which way was intended. 

A few possible solutions:
1. Update to the readme docs to also configure CalEPD/CMakeLists.txt for ESP32C3 devices, and clean up the comments in that file.
2. Add comments to the parallel epaper lines in comments in CalEPD/CMakeLists.txt.
3. Perhaps there's a way to handle what gets compiled with something like `#if CONFIG_IDF_TARGET_ESP32C3`
4. Or, instead of creating exceptions for specific boards, it looks like there's also a `CONFIG_IDF_TARGET_ARCH_RISCV` that could be used to flag which lines compile.

I'm pretty new to the C / C++ world so it took me a sec to figure out what was going on. I appreciate your work and effort with this library.

martinberlin, 2021-12-13:
Glad that it worked and you figured out how to fix it.

> Perhaps there's a way to handle what gets compiled with something like #if CONFIG_IDF_TARGET_ESP32C3

Yes that’s an idea that is on my mind since long. I’m not on holiday’s but in a few days will be back maybe you can help me figure out how to organize this in CMakeFiles, for sure there is a smart way to exclude this.
Other possibility would be to add in Kconfig the driver option:
SPI or Parallel (EPDiy) and if you select SPI it will automatically exclude that files plus being a bit more consistent on the fact that SPI pins should be defined.
Let’s give it a round of though and maybe if you have some time you can make a pull request?
Make sure to give a good read to the [esp-idf build system](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html)

martinberlin, 2021-12-13:
Forgot that “fixes ID” closes the ticket automatically;)

martinberlin, 2021-12-27:
Closing here since the issue is solved. Thanks for reporting.
Let's open a new one when we figure out what will be a more practical way to do this.

commit b0a488aa (2021-12-13): Fixes #52

Until we get a better idea on doing this automatically, parallel driver should be by default disabled
closed 2021-12-27 by martinberlin