martinberlin/cale-idf #57: ESP32-S3 build with error.

lanistor, 2022-06-14:
IDF: release 4.4
PC: MacOS 12.4
Target: ESP32-S3

Do i need some other configuration? 

Build end with lots of errors:
```
/Demo/cale-idf/components/epdiy/i2s_data_bus.c: In function 'i2s_int_hdl':
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:72:19: error: 'volatile union <anonymous>' has no member named 'out_done'; did you mean 'rx_done'?
   if (dev->int_st.out_done) {
                   ^~~~~~~~
                   rx_done
/Demo/cale-idf/components/epdiy/i2s_data_bus.c: In function 'i2s_switch_buffer':
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:93:50: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'out_link'
   while (i2s_is_busy() && dma_desc_addr() != I2S1.out_link.addr) {
                                                  ^
/Demo/cale-idf/components/epdiy/i2s_data_bus.c: In function 'i2s_start_line_output':
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:102:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_start = 0;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:103:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_reset = 1;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:104:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_fifo_reset = 1;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:105:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.rx_fifo_reset = 1;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:106:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_reset = 0;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:107:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_fifo_reset = 0;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:108:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.rx_fifo_reset = 0;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:109:6: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'out_link'
   dev->out_link.addr = dma_desc_addr();
      ^~
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:110:6: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'out_link'
   dev->out_link.start = 1;
      ^~
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:115:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_start = 1;
        ^~~~
        rx_conf
/Demo/cale-idf/components/epdiy/i2s_data_bus.c: In function 'i2s_bus_init':
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:131:21: error: 'I2S1O_DATA_OUT0_IDX' undeclared (first use in this function); did you mean 'I2S1O_WS_OUT_IDX'?
   int signal_base = I2S1O_DATA_OUT0_IDX;
                     ^~~~~~~~~~~~~~~~~~~
                     I2S1O_WS_OUT_IDX
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:131:21: note: each undeclared identifier is reported only once for each function it appears in
/Demo/cale-idf/components/epdiy/i2s_data_bus.c:145:8: error: 'i2s_dev_t' {aka 'volatile struct i2s_dev_s'} has no member named 'conf'; did you mean 'rx_conf'?
   dev->conf.tx_reset = 1;
...
```

With `main/CmakeList.txt`
```
idf_component_register(
    # CALE official ESP-IDF Firmware
    
    SRCS "cale.cpp"
    # SRCS "cale-7-color.cpp"

    # SRCS "demos/cale-sensor.cpp"
    # CALE with grayscale support (For now only plasticLogic)
    # Grayscale is also good to test parallel epaper with 16 gray levels
    SRCS "cale-grayscale.cpp"

    # Generic demos for touch displays (With an added layer or Gdew027w3T that comes with touch mounted)
    # SRCS "demos/demo-epaper.cpp"
    #SRCS "demos/demo-4-grays-7inch"
    
    # TOUCH examples. 1- Touch indepently from Epd class uses Gdew027w3   2- Touch implemented in Gdew027w3T class
    #SRCS "demos/demo-touch.cpp"
    #SRCS "demos/demo-touch-epd-implemented.cpp"
    #SRCS "demos/demo-touch-keyboard.cpp"

    # Generic demos for any displays
    #SRCS "demos/demo-sleep-clock.cpp"
    # SRCS "demo-sleep-clock-v2.cpp"
    #SRCS "demos/demo-fonts.cpp"

    # Demo to print 7 colors in Waveshare Acep epapers
    #SRCS "demos/demo-7-colors.cpp"

    # Demo only for plasticlogic.com epapers:
    #SRCS "demos/demo-epaper-plasticlogic.cpp"

    # Plasticlogic video demo: Please build it in www-image/video

    # Only parallel epapers (Driven by I2Sbus)
    SRCS "demos/parallel/demo-epaper-parallel.cpp"
    SRCS "demos/parallel/demo-epaper-parallel-partial.cpp"
    SRCS "demos/parallel/demo-touch-parallel.cpp"
    SRCS "demos/parallel/demo-touch-keyboard.cpp"
    INCLUDE_DIRS ".")

    # This is just to avoid that BitBank2 components like AnimatedGIF include Arduino.h
idf_build_set_property(COMPILE_OPTIONS "-D __LINUX__" APPEND) 

```

lanistor, 2022-06-14:
@martinberlin So CalEPD can drive parallel epaper without Epdiy?

lanistor, 2022-06-15:
Thanks, we are working on it.

martinberlin, 2022-06-15:
By the way if you would like to drive 8 or 16 datalines parallel both Waveshare and Goodisplay have a solution with IT8951. And for that you can use Lovyan GFX, there are examples here on this repository:
https://github.com/martinberlin/cale-idf/tree/feature/45-it8951
[Video demo](https://twitter.com/martinfasani/status/1536710420081397760)

About S3 please remember to adapt the host:
https://github.com/martinberlin/cale-idf/blob/feature/45-it8951/components/CalEPD/epdspi.cpp

For S3 you should also use: SPI2_HOST

lanistor, 2022-06-15:
Thanks for the suggestion。

martinberlin, 2022-07-09:
Any news here @lanistor ?
Did you succeed building this with S3?

lanistor, 2022-07-11:
Still trying, on the verge of giving up😂

martinberlin, 2022-07-11:
Strange. But did you follow my instructions so EPDiy component is not being compiled anymore?
If so, what is the next blocker?

martinberlin, 2022-07-26:
Closing here at the end of July since there is no more feedback since 2 weeks @lanistor 
Please keep the issue alive if you need support.

closed 2022-07-29 by martinberlin

martinberlin, 2022-07-30:
Updated CalEPD to support S3. Pull and give it a new try

lanistor, 2022-07-31:
CalEPD can drive parallel epaper with S3 now?

martinberlin, 2022-07-31:
No. Only EPDiy can but no S2 or S3 just ESP32 WRover with special PCB