martinberlin/cale-idf #61: L58Touch sometimes doesn't show x values between about 148 and 272.

laoshulaoshi, 2022-07-29:
**Describe the bug**
A column of my screen between the x values of about 148 and 272 sometimes doesn't show correct x values between those points.
-Touches can be detected anywhere on screen, including in that column.
-A stroke that begins somewhere in that column registers touches with y values correct, but the x values are always either about 148 or about 272.
-A stroke that begins outside of the column and crosses it may or may not register correct x values as it crosses the column. Strokes that are closer to horizontal seem to have a better chance of showing correct x values.
-When the x value does jump across the column in the middle of a stroke, event shown is 0 instead of 3.

**To Reproduce**
Lilygo T5-4.7 with capacitive touch cover. Initiate a touch between x = 148 and x = 272 to get an incorrect x, drag finger across that column from a point before or after to get possibly correct, possibly incorrect X values. Readout from a mostly-correct attempt:
02:35:24.484 -> X:125 Y:188 E:3
02:35:24.523 -> X:127 Y:189 E:3
02:35:24.562 -> X:129 Y:191 E:3
02:35:24.604 -> X:129 Y:191 E:3
02:35:24.653 -> X:132 Y:194 E:3
02:35:24.653 -> X:144 Y:198 E:3
02:35:24.694 -> X:162 Y:203 E:3
02:35:24.736 -> X:183 Y:209 E:3
02:35:24.774 -> X:203 Y:215 E:3
02:35:24.813 -> X:213 Y:218 E:3
02:35:24.861 -> X:235 Y:225 E:3
02:35:24.904 -> X:258 Y:232 E:0
02:35:26.472 -> X:274 Y:275 E:3
02:35:26.518 -> X:274 Y:275 E:3
02:35:26.560 -> X:274 Y:275 E:3
02:35:26.602 -> X:274 Y:275 E:3
02:35:26.642 -> X:274 Y:275 E:3
02:35:26.688 -> X:276 Y:275 E:3
02:35:26.688 -> X:279 Y:275 E:3
02:35:26.731 -> X:283 Y:275 E:3
02:35:26.772 -> X:287 Y:275 E:3
02:35:26.815 -> X:289 Y:275 E:3

Readout from a mostly incorrect attempt:
02:43:55.206 -> X:146 Y:474 E:3
02:43:55.246 -> X:147 Y:472 E:3
02:43:55.286 -> X:147 Y:472 E:3
02:43:55.325 -> X:147 Y:472 E:3
02:43:55.366 -> X:147 Y:472 E:3
02:43:55.405 -> X:147 Y:472 E:3
02:43:55.446 -> X:147 Y:472 E:0
02:43:55.486 -> X:273 Y:433 E:3
02:43:55.526 -> X:273 Y:423 E:3
02:43:55.566 -> X:273 Y:409 E:3
02:43:55.606 -> X:274 Y:392 E:3
02:43:55.646 -> X:274 Y:382 E:3
02:43:55.686 -> X:275 Y:366 E:3

**Expected behavior**
Touches should register the correct x and y values anywhere on the screen.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browser Firefox

martinberlin, 2022-07-29:
I could still not reproduce this on my end. Are you on the latest master branch of this repository?
You can also give it a try updating the components/FT6X36-IDF with the repository https://github.com/martinberlin/FT6X36-IDF

That one still does not have multi-touch merged. Will try to reproduce in the next days but I'm loaded with paid work. Please give it a try also with Lilygo example. This are the values that come via I2C so I'm not sure if the error is on my component.

laoshulaoshi, 2022-07-30:
Tried with Lilygo example and had the same issue, so it looks like a dodgy touchscreen on my end. Thanks for your time!

martinberlin, 2022-07-30:
No problem. I think is possible to remove it, but do it with car because you might brake the epaper

closed 2022-07-30 by martinberlin