martinberlin/cale-idf #74: FT6X36 ISR routine

jonsmirl, 2022-11-25:
The FT6X36 ISR needs to use xSemaphoreGiveFromISR() instead of xSemaphoreGive().

void IRAM_ATTR FT6X36::isr(void* arg)
{
	static BaseType_t xHigherPriorityTaskWoken;

	xHigherPriorityTaskWoken = pdFALSE;
	/* Un-block the interrupt processing task now */
    xSemaphoreGiveFromISR(TouchSemaphore, &xHigherPriorityTaskWoken);
	//xQueueSendFromISR(gpio_evt_queue, &gpio_num, NULL);
}

I'm trying to figure out how the gesture support works based on this code...
https://github.com/focaltech-systems/drivers-input-touchscreen-FTS_driver/blob/master/ft5x06.c

I want swipe left/right.

martinberlin, 2022-11-25:
If you find a way to make it better please add a merge request including that and optional software gesture.
But it should be optional. For example that part will only return a gesture if:
FT_GESTURE_ENABLE 

Is defined.

martinberlin, 2022-12-02:
@jonsmirl how is going?  Could you achieve something?

jonsmirl, 2022-12-02:
I discovered that LVGL already had a FT6X36 driver so I didn't need to port this one. I have switched onto the LVGL one.

Github won't let me attach, send me an email at gmail.com and I will send you the register definitions for the chip. The gestures are very easy to use. First you enable them. And then there is a register where you read the gesture ID. All of the recognition is done on the chip.

~~~
#define GESTURE_LEFT							0x20
#define GESTURE_RIGHT						0x21
#define GESTURE_UP		    					0x22
#define GESTURE_DOWN						0x23
#define GESTURE_DOUBLECLICK				0x24
#define GESTURE_O		    					0x30
#define GESTURE_W		    					0x31
#define GESTURE_M		   	 				0x32
#define GESTURE_E		    					0x33
#define GESTURE_L		    					0x44
#define GESTURE_S		    					0x46
#define GESTURE_V		    					0x54
#define GESTURE_Z		    					0x41
#define GESTURE_C                               0x34
~~~

[image]