martinberlin/cale-idf #81: Add two new epapers from Good-Display

martinberlin, 2023-02-09:
- [ ] 3.7" GDEQ037T31_416x240
- [ ] 400x200

gooddisplayepaper, 2023-02-11:
https://github.com/GoodDisplay/E-paper-Display-Library-of-GoodDisplay/tree/main/Monochrome_E-paper-Display/2.9inch_GDEM029E97_SSD1675
Here it is.

gooddisplayepaper, 2023-02-11:
https://github.com/GoodDisplay/E-paper-Display-Library-of-GoodDisplay/tree/main/Monochrome_E-paper-Display/3.7inch_GDEQ037T31_416x240

martinberlin, 2023-02-11:
Thanks @gooddisplayepaper started with the 2.9" and the fullscreen update is already working fine. But I'm getting an initial black line (Check the video) that still could not get rid of.
Then will try to get fast partial update working before starting with the 3.7"

martinberlin, 2023-02-13:
@gooddisplayepaper can you please check this example:

gooddisplayepaper, 2023-02-14:
Looks like a problematic example program. Outdated for some years. Last time tried GDEY029T94 Arduino example for it, full screen update wrong but the partial update worked (number 0 to 5)

gooddisplayepaper, 2023-02-14:
Also tried, partial update did not came. The program was like years ago long before I came to GD.
You were trying based on SSD1675B GDEY0213B73 class ?

martinberlin, 2023-03-16:
Closing here. Reopening when @gooddisplayepaper enginners publish some partial update examples

martinberlin, 2023-05-17:
Good Display updated their example for [GDEQ037T31](https://github.com/GoodDisplay/E-paper-Display-Library-of-GoodDisplay/tree/main/Monochrome_E-paper-Display/3.7inch_GDEQ037T31_416x240) Adding 4 grays:

https://github.com/GoodDisplay/E-paper-Display-Library-of-GoodDisplay/blob/main/Monochrome_E-paper-Display/3.7inch_UC8253_GDEY037T03_416x240/HARDWARE/EPD/Display_EPD_W21.c

Reopening till this part is implemented and tested

martinberlin, 2024-03-09:
Not really have the time to add more SPI epapers here

closed 2024-03-09 by martinberlin