martinberlin/cale-idf #83: Fail to find the reason，Why come out an extra black line near the right boarder in Gdey029T94's screan

hzlijianjun, 2023-02-16:
Hi, Honored cale-idf‘s author
    When I test cale-idf with a simple example in my e-paper Gdey029T94，I found an extra black line near the right boarder in the screan。Switching to Arduino，There‘s no occured the problem as I run some GxEPD2 examples。Can you help to figure out the reason。

[image]

Hardware/Software:
1）Esp32 DevKitc-V4
2）Gdey029T94
3）cale-idf，master

`
...
#include "goodisplay/gdey029T94.h"
// After that check WiKi for examples
EpdSpi io;
Gdey029T94 display(io);

...

extern "C" void app_main(void){
   printf("CalEPD version: %s\n", CALEPD_VERSION);

   // Test Epd class
   display.init(false);
   display.update();

   printf("display: We are done with the demo\n");
}
`

martinberlin, 2023-02-17:
Hi hzlijianjun, thanks for your report. This is one of the latests display's added
#define GDEY029T94_WIDTH 128
#define GDEY029T94_HEIGHT 296

Thanks I will take a look on this does not look hard to solve. Probably this display needs an additional 128 line that I'm missing.

martinberlin, 2023-02-17:
Please pull latest changes and close it after testing that this bug is corrected. Thanks!

hzlijianjun, 2023-02-19:
well, well! I was glad to pull the code and have done a test. which was functioning correctly without that problem again.

closed 2023-02-19 by hzlijianjun