martinberlin/cale-idf #85: Add waveshare 3.7in epaper display

MingyaoLiu, 2023-02-24:
**Is your feature request related to a problem? Please describe.**
I got some 3.7 inch waveshare epaper display that uses 24 pin fpc, and I tried their arduino demo code and it works fine (supports partial refresh). It works with the existing waveshare HAT, I tried it with my 2.9 inch v2 HAT it works, I also built a simple pcb following their old 3.3v only HAT design and it also works fine.
**Describe the solution you'd like**
I would like to build a project using this board and esp32c3 (or esp32c6). I have tried esp32c3 with this board on arduino, I couldn't figure out how to make greyscale work, but black and white seems to work fine. partial refresh also works fine.

**Describe what epaper display you want to use**
waveshare 3.7 inch epaper
https://www.waveshare.com/product/displays/e-paper/epaper-2/3.7inch-e-paper.htm

**Additional context**
The arduino demo from waveshare : https://www.waveshare.com/wiki/3.7inch_e-Paper_HAT_Manual#Resources

I would love to help with this, I'm just not knowledgeable enough with how ESP32-IDF works (1 day experience). I can help test this.

martinberlin, 2023-02-24:
Hi Ming,
The problem with this display is that I’ve never had one in my hand and will probably never get one since I mostly use Goodisplay epapers.
The easier and faster solution is that you make this yourself: just copy one of the existing classes, rename it everywhere the same as you can see in other displays, and implement the new class for yours, following examples from the display producer. But of course for this first you need to speak some time with IDF and get familiar with it.

The lazy take is that you send a display to me and I will do what I can for you, if I manage to get it working I will keep it as a payment for my time and you get your class that will be also open source for the rest. But I can’t guarantee that everything will work as you expect,
Partial update is not easy to get it working and I cannot even guarantee for the ones that I implemented myself.

Feel free to make your decision and let me know.

MingyaoLiu, 2023-02-24:
Cool, let me get my hands dirty on this for now, and if I can't handle it we can figure something out. Thanks.

MingyaoLiu, 2023-02-25:
@martinberlin I see 2 repos, one called cale-idf, another one is calepd, which one should I try to add the waveshare 3.7 inch epaper ? I don't quite know which one is proper for this.

martinberlin, 2023-02-25:
I do it like this: 
Develop and add new epapers in cale-idf that has the components directly there (without being linked as git submodules)
After testing and checking that everything works, then I copy the update to CalEPD.

So if you add a new epaper, just fork cale-idf anf work from there, then when it works you can make a **Merge request** and I will review it.
Please mind the defines to set WIDTH / HEIGHT of your display and try to use the existing nomenclatures DISPLAYMODEL_WIDTH, etc. New models also have a function called setMonoMode(boolean) on false is 4 gray mode.
4 gray uses 2 buffers like having 2 layers in the display. But do not care about 4 gray at the beginning just focus in getting monochrome working.
Check to send right SPI **_wakeUp** commands, to send the right framebuffer that the EPD expects and finally the right update commands and sleep. That should be mostly it at the beginning. All the rest like 4 grays and partial update can be done after that initial steps are done. Good luck!

MingyaoLiu, 2023-02-25:
Got it, I will fork cale-idf and work from there, thanks.

martinberlin, 2023-03-16:
Any updates on this one? 
Did you achieved a refresh?

martinberlin, 2023-05-02:
Hello @MingyaoLiu please keep me updated about the issues / challenges you face with this. If there is not an update in the end of May I will close the issue since it does not make much sense to keep stuff open when there is no progress.

MingyaoLiu, 2023-05-02:
Hey I haven't had any time to implement this, I will be looking into it again probably in July, you can close this and I will do a fork if I work on this again.

Thanks

martinberlin, 2023-05-02:
Thanks for the update. Just reopen an issue with display model and I will try to find time to help you out with this

closed 2023-05-02 by martinberlin