martinberlin/cale-idf #86: To Fasani corp and all CALE-IDF supportors regarding 7.5 inch #GDEW075T7 / #GDEY0757T/ #GDEY075Z08 and #GDEW075T8

gooddisplayepaper, 2023-03-16:
**Describe the bug**
Regarding 7.5 inch #GDEW075T7 / #GDEY0757T/ #GDEY075Z08 and #GDEW075T8,

For more info, plz check the tweets:
https://twitter.com/Frrrrrrreddie/status/1636198925593092102

#GDEW075T7& #GDEY075T7 
- this is EOL display and no more stock from 2022. But Freddie (de facto that's me) had earlier today on CALE-IDF tested with #GDEY075T7 this currently popular item from us. Worked very well.
See the link for the #GDEY075T7 -
https://www.buy-lcd.com/products/75-inch-large-e-ink-screen-spi-electronic-paper-display-gdey075t7
So, what about put notes and update so everyone will then know oh, I could simply use this one to try out!

#Also for GDEY075T7 trying out 4Gray demo, flash ok but won't get it work and I spotted endless serial print...

#GDEY075Z08
- this is not yet added and we so far see no 7.5 inch BWR (BWY yellow ones btw all suppliers will not do this, so we have some left but no need to bother Cale-IDF.) Will send one pcs to dear Martin and also do sth based on what we already have.
- Not good at pull request so might offer the demo code here; 
URL: 
https://www.buy-lcd.com/products/e-ink-module-800x480-75-inch-red-tri-color-electronic-paper-display-gdey075z08

#GDEW075T8
- this is also EOL display, but luckily we have some left in stock. 
- And I can give good price (yes, this only takes effect you contact me because I said it not my colleagues)
so why not we support some makers for pretty nice and small projects? 
But testing it I found compile and flash ok, but won't get me a refreshed EPD.
[image]

We will do some work on our side meanwhile shall have Martin with a solid panel to see what's really going on.

martinberlin, 2023-03-16:
Great I’m slowly getting fit and will answer this in detail. Two things from the top of my head:
GDEY075Z08 true never had one of that ones so we could add it. I still have the old dinosaur of the gold/black somewhere and still refreshes

About https://github.com/martinberlin/cale-idf/wiki/Model-gdew075T7.h#4-gray-version did you try the 4 gray class? Because this one’s have a quite big buffer to keep everything in RAM.
Anyways the 4 grays in this model never got it 100% right like in the 400x300 version (ultrachip)
About the Wiki page titles: I’m finding it hard to find epaper models. What do you think of renaming them like this:
Model name / width x height / BR (tint colors if other than black)
Example:

gdew075T7 800x480

gooddisplayepaper, 2023-03-20:
> Great I’m slowly getting fit and will answer this in detail. Two things from the top of my head: GDEY075Z08 true never had one of that ones so we could add it. I still have the old dinosaur of the gold/black somewhere and still refreshes
> 
> About gdew075T7.h did you try the 4 [gray class](https://github.com/martinberlin/cale-idf/wiki/gdew075T7-800x480#4-gray-version)? Because this one’s have a quite big buffer to keep 4 buffers in RAM and make a function like in new displays setMonoMode(bool). With 3 buffers I refer to:
> 
> 1. Monochrome buffer
> 2. RAM1 4 grays
> 3. RAM2 4 grays
> 
> for the 4 grays you need to send usually to the 2 RAMs of the display and first put it in the 4 gray Waveform mode.
> 
> Anyways the 4 grays in this model never got it 100% right like in the 400x300 version (ultrachip) About the Wiki page titles: I’m finding it hard to find epaper models. What do you think of renaming them like this: Model name / width x height / BR (tint colors if other than black) Example:
> 
> gdew075T7 800x480

Tried with GDEY075T7 using demo4gray and also demograys. No luck with EPD. see the video
Will then need to get you a solid sample.
https://twitter.com/i/status/1637730544057466880

closed 2023-08-01 by martinberlin