martinberlin/cale-idf #89: Wave12I48RB Build Errors

Ryan-Finn, 2023-04-01:
Hello, I'm working with the Waveshare 12.48in RB Module (https://www.waveshare.com/12.48inch-e-paper-module-b.htm) and the Waveshare ESP32 Driver Board (https://www.waveshare.com/e-paper-esp32-driver-board.htm). I tried Waveshare's demo examples but was unable to get any of it to work, so I decided to look elsewhere, which lead me here.

I installed ESP-IDF v4.4.4 on my Windows 10 machine as outlined in the Wiki (https://github.com/martinberlin/cale-idf/wiki/Installing-Espressif-IDF-toolchain-on-Windows). Because I am using VSCode I also installed the ESP-IDF extension (https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md) with the "Use Existing Setup" setup mode.

Now following this Wiki (https://github.com/martinberlin/cale-idf/wiki/Building-CalEPD-with-Espressif-IDF), I cloned this repo and in VSC I edited main/CMakeLists.txt to:
```
idf_component_register(
    # . . .
    SRCS "demos/demo-fonts.cpp"
    # . . .
    INCLUDE_DIRS ".")

    # This is just to avoid that BitBank2 components include Arduino.h
idf_build_set_property(COMPILE_OPTIONS "-D __LINUX__" APPEND)
```

I then set the root sdkconfig file to be the same as config-examples/sdkconfig_Wave12I48

Opening main/demos/demo-fonts.cpp I run into the first problem:
[image]

It would seem several header files can't be found in the project repo. This may be the root of my build issue, but I'm not sure honestly because the compiler doesn't actually fail until step 1095/1097. I do get hundreds of [-Wmissing-field-initializers] warnings beforehand though:
[image]

I build using the Ctrl + E, B commands as part of the ESP-IDF VSC extension.

Any guidance would be greatly appreciated. I'm a complete newbie at this, so I could be missing something completely obvious.

martinberlin, 2023-04-01:
Hi Ryan,
Make sure that the class and dependencies like **epd4spi.cpp** are not commented out in the component CMakeFiles first:

https://github.com/martinberlin/cale-idf/blob/master/components/CalEPD/CMakeLists.txt

Ryan-Finn, 2023-04-01:
Okay, that definitely helped. I'm now able to build. After setting my flash mode to UART I get the following output when flashing:
```
 *  Executing task: C:/Espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe C:\Espressif\frameworks\esp-idf-v4.4.4\components\esptool_py\esptool\esptool.py -p COM3 -b 460800 --before default_reset --after hard_reset --chip esp32 write_flash --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader/bootloader.bin 0x10000 cale-epd.bin 0x8000 partition_table/partition-table.bin 

esptool.py v3.3.2
Serial port COM3
Connecting....
Chip is ESP32-D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 78:e3:6d:13:98:54
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00001000 to 0x00007fff...
Flash will be erased from 0x00010000 to 0x0004afff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash params set to 0x0220
Compressed 25520 bytes to 16000...
Wrote 25520 bytes (16000 compressed) at 0x00001000 in 0.7 seconds (effective 289.8 kbit/s)...
Hash of data verified.
Compressed 238800 bytes to 125233...
Wrote 238800 bytes (125233 compressed) at 0x00010000 in 3.3 seconds (effective 577.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 486.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

Which looks good I think, no errors here. However, nothing happens on my board. I'm connected to the right COM, and looking at the serial monitor I see:
[image]
[image]
[image]
[image]

Sorry for all the screenshots. After the "Rebooting..." it just repeats the above text over and over. There's something happening when the Wave12I48RB::fillScreen function gets invoked during the init that's causing this discriminator 2 error, but I'm not sure what.

martinberlin, 2023-04-02:
1st question will be: Did you made the hardware modification I indicated on the wiki?

This needs an ESP32 with external RAM. Otherwise it won’t work. You cannot possibly allocate two big buffers and have GFX without PSRAM and this has to be enabled in menuconfig too.

Please read https://github.com/martinberlin/cale-idf/wiki/wave12i48-1304x984-EOL#remarks

martinberlin, 2023-05-17:
Closing this since didn’t got any more feedback and I’m quite sure that there is a missing hardware hack to make it work with this display (please write me a line if you need me to reopen or you cannot make it work)

closed 2023-05-17 by martinberlin