martinberlin/cale-idf #91: Mirrored text

Johboh, 2023-05-17:
Hi!
I have a issue where the text I'm printing is mirrored, and I have a bit of a struggle trying to find out why.
This is the code (from [small-display.cpp](https://github.com/martinberlin/cale-idf/blob/master/main/demos/small/small-display.cpp))
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <stdio.h>
#include "color/gdeh0154z90.h"

EpdSpi io;
Gdeh0154z90 display(io);

#include <Fonts/FreeMono18pt7b.h>

extern "C" {
void app_main();
}

void app_main(void) {
  display.init(false);
  vTaskDelay(1000 / portTICK_RATE_MS);

  display.setRotation(2); // without this, text is upside down (but still mirrored)
  display.fillScreen(EPD_WHITE);
  display.setTextColor(EPD_BLACK);
  display.setFont(&FreeMono18pt7b);
  display.setCursor(10, 20);
  display.print("Hello!");
  display.update();
}
```

This is for when using this component (@master branch) together with [Adafruit GFX IDF@master](https://github.com/martinberlin/Adafruit-GFX-Library-ESP-IDF) in an ESP-IDF (v.4.4.4) + Visual Studio Code project.
[image]

However, If I setup a platform I/O Arduino project and uses [zinggjm/GxEPD](https://github.com/ZinggJM/GxEPD) with the same code (and same device and display), the text is correct. Note that in this case I don't need to set the rotation.
```
  display.fillScreen(GxEPD_WHITE);
  display.setTextColor(GxEPD_BLACK);
  display.setFont(&FreeMono18pt7b);
  display.setCursor(10, 20);
  display.print("Hello!");
  display.update();
```
[image]

I'm most probably missing something here. What could cause this?

martinberlin, 2023-05-17:
Maybe I didn't test rotation right. Will take a look but I don't have one of this with 2 colors. 
What you can try and fix it yourself since this was a external contribution that I approved but never had a display to test.

Just try to invert X: (But also test if drawing it Red makes the same effect, notice there are 2 buffers)
https://github.com/martinberlin/cale-idf/blob/master/components/CalEPD/models/color/gdeh0154z90.cpp#L100 

```C
    for (uint16_t y = 1; y <= GDEH0154Z90_HEIGHT; y++)
    {
       //for (uint16_t x = 1; x <= xLineBytes; x++)  ORIGINAL LINE Change it to:
       // NOT sure if it's x because of the rotation. If is not leave it like this and invert Y direction following this
       for (uint16_t x = xLineBytes; x > 0 ; x--)
        {
            uint8_t data = i < sizeof(_black_buffer) ? _black_buffer[i] : GDEH0154Z90_8PIX_WHITE;
            x1buf[x - 1] = data;
            if (x == xLineBytes)
            { // Flush the X line buffer to SPI
                IO.data(x1buf, sizeof(x1buf));
            }
            ++i;
        }
    }

```

If that works please test all rotation modes to see it works as expected (Write 2 lines one in each color)
And then please make a Merge Request so you contribute to fix the issue. Thanks!

Johboh, 2023-05-18:
@martinberlin I cherry picked your [commit](https://github.com/martinberlin/cale-idf/commit/c336c7c81f45e69a4dcd72d30b68463747b2c769) and now the text is correct, thanks! I also verified black and red colors as well as that all rotations where correct.

closed 2023-05-18 by Johboh

martinberlin, 2023-05-18:
Thanks for the test. Will merge it into master soon!
Mind that to use this as a component you need to use [CalEPD](https://github.com/martinberlin/CalEPD) that is in a different repository. Is easier and lighter. Moving this update there also quite soon

Johboh, 2023-05-18:
Yes I'm using [CalEPD](https://github.com/martinberlin/CalEPD) and [Adafruit GFX IDF](https://github.com/martinberlin/Adafruit-GFX-Library-ESP-IDF) as components in the project.

martinberlin, 2023-05-18:
Latest this weekend develop is merged in master here and then I copy the contents to CalEPD and make a new release