martinberlin/cale-idf #97: Partial refresh failure on GDEY075T7

XAG-labs, 2023-11-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Specify what Model/Classname of epaper you are using.
Write meaningful steps to reproduce the behavior:
1. Compile the SRC file ***.cup for target ESP32** (S2?)
2. Does it build?
3. Where it hangs? If so please provide the serial logs related to this
4. If it’s a rendering issue please upload photos of how it renders on your display

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

martinberlin, 2023-11-20:
Sadly I don't have anymore one of such displays at hand @XAG-labs . What you can try is to send first the whole area to white, execute the partial commands and then try again with the partial buffer.
If that works you can make a Pull request and contribute to this project. Another alternative is to check how is done in GxEPD /(Or GxEPD2) and replicate those commands in the class. 
Partial updates in SPI displays is always hard to get working 100% correctly and I really stopped trying since I don't have the time anymore to have this battles when I have many other important matters to resolve

XAG-labs, 2023-11-21:
> Sadly I don't have anymore one of such displays at hand @XAG-labs . What you can try is to send first the whole area to white, execute the partial commands and then try again with the partial [...]

Thank you Mr. Martin for the reply. 
I will definitely try both the solutions that you have provided and I will update you regarding them

martinberlin, 2023-11-24:
Hello @XAG-labs 
Could you advance trying some way to fix this partial refresh issue?
Just tell if I can help you at some point.

XAG-labs, 2023-11-24:
> Hello @XAG-labs Could you advance trying some way to fix this partial refresh issue? Just tell if I can help you at some point.

Hello Mr. Martin,
The first solution which you have provided which is, "send first the whole area to white, execute the partial commands and then try again with the partial buffer." did not work unfortunately.
The second solution should work because the partial refresh is working absolutely fine in Arduino using the GxEPD2 library. 
Right now as I'm busy with other work, I couldn't start with the second solution. 
I will definitely contact you if I need any help.

Thank you,
Shreyas

martinberlin, 2024-03-09:
Closing here since it's 3 months of inactivity. @shREYas-310 please open another issue about it if there is really an interest to help solving it

closed 2024-03-09 by martinberlin