martinberlin/cale-idf #103: Issue compiling for esp32-c3

Jim-Foo, 2025-03-10:
**Describe the bug**
A clear and concise description of what the bug is.

When trying to compile for an esp32-c3 I get an error that SPI3_HOST is not defined, referring to the macro at the top of the file CalEPD/epd4spi.cpp. Apparently the esp32-c3 doesn't have  SPI3. I changed it to SPI2 and it compiled and works. Another error was at CalEPD/epd.cpp:142:5. "error: 'uint' was not declared". I changed it to uint16_t.

**To Reproduce**
Specify what Model/Classname of epaper you are using.
Write meaningful steps to reproduce the behavior:
1. Compile the SRC file ***.cup for target ESP32** (S2?)
2. Does it build?
3. Where it hangs? If so please provide the serial logs related to this
4. If it’s a rendering issue please upload photos of how it renders on your display

The epaper I'm using is 2.13 inch made by weact and uses a SSD1680 chip. It seems to work ok with the "goodisplay/gdey0213b74.h" header. The issue is with the particular mcu not the display.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

jpg0, 2025-05-08:
I have this too for ESP32-S2 for the uint problem (but not the SPI3 one).

martinberlin, 2025-05-10:
Hey guys, thanks for the report. This is open source and I really don't have time at the moment to maintain this. 

Will take over in the next month. Please be nice and make a pull request with those two small fixes and I will review it and merge it. Thanks in advance!